ArduPilot/ardupilot #9164: Ardupilot 3.9 RPM not working on ChibiOS firmware (NuttX is OK)

suraqis, 2018-08-09:
After firmware update on Pixhawk 2.1 from 3.8 to 3.9 ChibiOS without changing any parameters, RPM sensor didn't work at all. After changing firmware to 3.9 NuttX it works well. 

RPM_TYPE = 1 (PX4-PWM)
RPM_PIN = 54 (AUX5)
 

### Bug report

**Issue details**

_Please describe the problem_

**Version**
Ardupilot 3.9 ChibiOS

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed Wing

**Hardware type**
Pixhawk 2.1

**Logs**
No data can be logged regarding the issue.

rmackay9, 2018-08-09:
I think this is a known issue but @peterbarker is working on a fix.  I think basically the gpio interrupt feature isn't complete on ChibiOS quite yet but will be soon and you can probably expect it to appear in a point release of plane (I'm guessing though)

peterbarker, 2018-08-14:
@suraqis It has been PR'd to master

peterbarker, 2018-08-26:
It's been merged to master.  Good luck!

closed 2018-08-26 by peterbarker