ArduPilot/ardupilot #9170: about AP_Beacon_Pozyx

icyore, 2018-08-10:
### Bug report

**Issue details**

I think the following code is wrong.
 set_vehicle_position(veh_pos, position_error); 
should be 
set_vehicle_position(veh_pos, position_error /1000.f);
because 
void AP_Beacon_Backend::set_vehicle_position(const Vector3f& pos, float accuracy_estimate)
in libraries/AP_Beacon/AP_Beacon_Backend.cpp
and 
convert int16_t position_error to float accuracy_estimate directly is wrong

What's more, if the pozyx system works，should the LED of pixhawk  turn green？
**Version**
dev3.6

**Platform**
[] All

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Pixhawk

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-08-17:
I don't think the LED turns green because it required the GPS status to be 3D lock rather than simply requiring the EKF position estimate to be good.  I think it would be a good idea to make the LED turn green when the EKF position estimate is good (i.e. remove the GPS requirement).

icyore, 2018-08-29:
I think it is ok to send fakeGPS message to pixhawk. Is it better to send four distance to beacon as part of EKF. Why?

rmackay9, 2018-08-29:
@icyore, because with beacons, we get at least 3 distances sent into the EKF and each of them can be useful.  Even 1 beacon will improve the EKF's estimation.  I think the EKF can do a better job of combining each distance than an external system can of calculating the 3d position.

icyore, 2018-08-29:
When I use the beacon, it shows the "EKF3 IMU%u stopped aiding" in MP. I don't know why. What's more, How can I improve the accuracy of position. Because there are 1 meter between BCN.PosX and XKF1.PN

IamPete1, 2021-08-30:
@icyore @rmackay9 It looks to me that the error is unit-less convarance.

https://github.com/pozyxLabs/Pozyx-Arduino-library/blob/fe81ec7303047781250974827845ba8a653de6d4/Pozyx_definitions.h#L80

In which case we can close this, the LED colors should change, but were have other open issues for that.

rmackay9, 2021-08-31:
@IamPete1, I think we could still have the scaling wrong though.  The result could be subtle and would explain the lack of accuracy reported here.  Sadly I don't have this system anymore so I can't say if it is really a problem.