ArduPilot/ardupilot #9172: Rover: add support for stepper motors

rmackay9, 2018-08-11:
Rover-3.5 supports Balance Bots (aka segway like vehicles) but we could achieve better performance by adding support for [stepper motors like these from pololu.com](https://www.pololu.com/category/87/stepper-motors).  Performance is improved because stepper motors do not have the slight play in the motor shaft that brushless motors have.

This would involve:

- adding a new MOT_PWM_TYPE to the [AP_MotorsUGV class](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/AP_MotorsUGV.cpp).
- extending [AP_HAL/RC_Output](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/RCOutput.h) class to support outputting NMEA messages (many stepper motors accept a serial input in NMEA format) or something close to a duty cycle where toggling a pin causes the motor to step forward (or backwards) one step.

One additional potential advantages of using stepper motors is we may not require wheel encoders because we can assume that whatever we output to the motors/motor driver is actually executed (I'm unsure on this last point).

IamPete1, 2018-08-11:
This would also be excellent for antenna tracker, much cheaper than big servos. I think it can be done as a extension to the DC motor with reversing relay, it just needs to output a faster rate of pulses rather than PWM. 

Would take some addition logic and a parameter defining the deg/step to keep track of the position. For antenna tracker it would also be use full to have some sort of homing switch like a 3D printer to stop it from crashing into its self.

Jdunkle, 2018-10-19:
Anybody using stepper motors with pixhalk ballence bot. I am converting one off my pixhalk ball bot from brushed to stepper and I'm looking for any sugestions.

faerietree, 2022-01-01:
Me fool used steppers for my rover. Now I’m lost