ArduPilot/ardupilot #9192: Plane: use acro roll limit in roll PID in ACRO mode

tridge, 2018-08-14:
we should override the roll rate limit to match the ACRO_ROLL_RATE when in ACRO mode. See 
https://discuss.ardupilot.org/t/chibios-port-of-ardupilot/20491/70?u=tridge

Naterater, 2018-08-19:
Where is RLL2SRV_RMAX used? It was my understanding that automatic missions, etc. did immediate stepped roll commands instead of ramped changes that were clamped by this parameter. If we just removed RLL2SRV_RMAX, wouldn't that accomplish the same thing?

Hwurzburg, 2019-04-30:
This still is popping up in the RCG threads....personally, I think that ACRO rates should apply (ie override the x_RMAXs) to both ACRO and STABILIZE since they are both aerobatic modes (ie 2D and 3D in other FC parlance)....

IamPete1, 2019-05-01:
i have done a fix for the roll limit, it seems that the pitch limit already works as it should. @Hwurzburg Is this in line with what is being reported?

Hwurzburg, 2019-05-01:
Not sure, most complaints were about roll since that is always much faster than pitch in a plane due to axis stability differences....does this fix also allow stability mode to roll as fast as ACRO?

IamPete1, 2019-05-01:
this is just a acro mode bug fix to match the param descriptions currently, stabilize mode would be a change of behavior, not that we shouldn't do that too but i think should be a different PR

IamPete1, 2019-05-06:
PR merged, acro roll rate limit should now over ride RLL2SRV_RMAX in acro mode

closed 2019-05-06 by IamPete1