ArduPilot/ardupilot #9208: Do_change_speed GS option inop

jace26, 2018-08-16:
**Issue details**
While trying to make adjustments for a future flight I simulated speed changes to slow down our approaches. However, Do_change_speed commands with column one set to 1 (for GS instead of AS) do not result in any speed change. I only found this issue brought up before in reference to rover issues.

**Version**
3.8.5beta1
3.9.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octo-quadplane

**Hardware type**
pixhawk 2.1 - simulated only so far

**Logs**
tlog with two flights b2b, first with GS speed change, second with AS 
also add ability to upload tlogs and wp files..
tlog:
https://drive.google.com/open?id=1Dh5cbbL980PmefoNyl0eAH_1N8otsTXy
wp1
https://drive.google.com/open?id=1E-EVjczR-9ZoirV1UpqZL6BtFD8hIHzb
wp2
https://drive.google.com/open?id=1Dhak5N_pbGyLJ9jMbAA0cJpPzbP5mDpS

WickedShell, 2018-08-16:
Plane does not currently support any ground speed based modes, we only have a parameter to enforce a minimum ground speed, rather then maintaining a desired ground speed.

It's a valid feature to add, although not one I can volunteer on at this point.

jace26, 2018-08-16:
I can see how for normal airplanes this could be an infrequently used feature, but for a vtol plane, the GS is obviously more important than AS in the VTOL phase.

WickedShell, 2018-08-16:
The airspeed is only taken into account when doing a transition or in normal flight I believe. I know it doesn't have an effect during qloiter or similar.

jace26, 2018-08-17:
I just discovered another similar issue, for both AS and Throttle changes that I'll present in a new issue.