ArduPilot/ardupilot #9218: Plane: Do_change_speed => cannot return to default

jace26, 2018-08-17:
When using the Do_Change_Speed command during an auto flight there is no way to revert control of the speed/throttle to automatic control. For a AS change this isn't to terrible, as you can just create a Do_Change_Speed command to return to the default AS for the aircraft, but with throttle changes this is a huge issue. If I command a throttle change for whatever reason, I am from that point on responsible for throttle output through all phases of flight, until the FC resets. It appears that reloading WPs, changing modes out of auto, and disarming will not reset the AS or Throttle control to default. Another major problem here is that an RTL will not take control over this either so I've simulated driving this into the ground during an RTL with a low throttle commanded by a do_speed_change  

This has all been done in simulator runs.

I propose a change where by setting either of these to 999 with the do_change_speed would return them to auto control. And an RTL should set throttle and AS to default, as should a disarm. 

**Version**
3.9 
3.8.51

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2018-08-19:
On Fri, 17 Aug 2018, Jace McCown wrote:

> I propose a change where by setting either of these to 999 with the
> do_change_speed would return them to auto control And an RTL should set

The DO_CHANGE_SPEED message includes a field indicating whether the change 
is relative or absolute (which we don't currently use).  I'm assuming that 
would basically be "over what the aircraft would otherwise say" 
(multiplicative with throttle-nudging, I guess).  A 0% relative would 
remove the effects of any previous DO_CHANGE_SPEED,

> throttle and AS to default, as should a disarm.

Trickier - if you're relying on the speed-boost to get you home, this 
could be a problem.

jace26, 2018-08-21:
That sounds like an interesting possibility. I'd love to hear more on putting this into effect. Unfortunately I'mat the mercy of those who are capable of doing this as it is beyond me currently.

peterbarker, 2018-08-28:
This was discussed on the devcall today.  The main debate was how to interpret "relative" - as in relative to current throttle or relative to base desired speed (e.g. waypoint navigation speed), or relative to your current speed - that is whether sending a 115% relative value followed by a 110% relative throttle value would result in a 26.5% increase in speed or a 10% increase in speed.  We decided on the latter, based on mavlink being a lossy protocol so it's hard to tell if the command has had an effect or not (the ack might be lost).

I've said I'll put a PR together based on that assumption.

rmackay9, 2018-08-28:
@peterbarker, I disagree with that decision that it should be relative to a base desired speed so not happy to have that in Copter or Rover.

WickedShell, 2018-08-28:
I'd also support there is a lot of value to a GCS with gamepad controls to be able to do a relative to current speed adjustment without needing to know all the vehicle parameters. For this you need the relative control which is much more intuitive. So I'm very strongly with @rmackay9 on this one.

I offered on the call (but I suspect my microphone wasn't working at the time) to sanity check the desired meaning of the field with the MAVLink call so we don't get another divergence. There is also merit to a 3rd value for the field of "relative to base" which peter suggested on the call.

jace26, 2018-08-28:
From a functional perspective I can see both sides. For something where you would be controlling manually or modifying automatic response while the vehicle is operating relative to current makes sense, much like adjusting a cruise control on a car up or down. But, foe wholly pre-planned missions (what I am doing) modifying relative to a base speed seems to make more sense as does defining a specific speed which is what I currently do, so long as I can return to the base speed (which in my case is my safe, known speed) for RTL or at the end of maneuvering. Throttle changes are similar, and what actually prompted me to bring this up in the first place, which also makes me want to clarify something.

jace26, 2018-08-28:
Maybe we also need another RTL parameter which defines how speed will be handled during an RTL similar to how altitude works? You could then set an RTL speed, have it continue at current speed, or go back to the base speed.

WickedShell, 2018-08-29:
Lets back up one here: Airspeed changes are strictly in meters per second, there is no way to apply a percentage to it. Percentage is only for the throttle channel per the message spec. So what I want is a way to send 1 m/s as a relative change that says "just increase target airspeed by 1 m/s from whatever it is now". (Note the special case of avoiding -1.0 for relative the opposite direction). On the throttle channel it's then expressable as raise setpoint by 10% of overall throttle range so say it was at 50% throttle it would move to 60%.

Message pasted for sanity below:
```
      <entry value="178" name="MAV_CMD_DO_CHANGE_SPEED">
        <description>Change speed and/or throttle set points.</description>
        <param index="1">Speed type (0=Airspeed, 1=Ground Speed)</param>
        <param index="2">Speed  (m/s, -1 indicates no change)</param>
        <param index="3">Throttle  ( Percent, -1 indicates no change)</param>
        <param index="4">absolute or relative [0,1]</param>
        <param index="5">Empty</param>
        <param index="6">Empty</param>
        <param index="7">Empty</param>
      </entry>
```