ArduPilot/ardupilot #9242: Quadplane automatic transition using VTOL_TAKEOFF

merbb, 2018-08-21:
### Issue details

So i want to make automatic transition in code from VTOL to plane. I want to add my code to Quadplane.ccp line 2088 "start a VTOL takeoff". To make it safe and stabilised I want to start VTOL for about 3-4 m and then using WVANE_GAIN to yaw face into the wind and then start transition during flying towards the wind. After that my plane will go to the first WP. 

Do i think correctly? 
I need some help in coding ! :) thanks for help in advance

Sorry for my english, it's not my native language.

Version
Literally all versions since the quadplane support was added. Lately 3.8.5 beta 1.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

PH2.1

IamPete1, 2018-08-21:
you can do this using mission commands, something like DO_VTOL_TAKEOFF this will take off to a set altitude, if you then put in a delay it will wait at that altitude. You can have it wait for 10 or 20 seconds for the wind vanning to kick in, then use a DO_VTOl_TRANSTION command, then the rest of you motion

merbb, 2018-08-21:
Thanks for response. You are talking about doing it in MissionPlanner not in code? Something like this one?
[image]

IamPete1, 2018-08-21:
yeah, exactly, you need a vtol transition before and after all your way points or it will just stay in quad mode

merbb, 2018-08-21:
I understand. But I need to make strictly in code. 

Like this for example

  start a VTOL takeoff
 */
bool QuadPlane::do_vtol_takeoff(const AP_Mission::Mission_Command& cmd)
{
    if (!setup()) {
        return false;
    }
    plane.set_next_WP(cmd.content.location);
    plane.next_WP_loc.alt = plane.current_loc.alt + cmd.content.location.alt;
    throttle_wait = false;

    // set target to current position
    loiter_nav->init_target();

    // initialize vertical speed and acceleration
    pos_control->set_speed_z(-pilot_velocity_z_max, pilot_velocity_z_max);
    pos_control->set_accel_z(pilot_accel_z);

    // initialise position and desired velocity
    set_alt_target_current();
    pos_control->set_desired_velocity_z(inertial_nav.get_velocity_z());
    
    // also update nav_controller for status output
    plane.nav_controller->update_waypoint(plane.prev_WP_loc, plane.next_WP_loc);
    return true;
}
This is example code from quadplane.cc

IamPete1, 2018-08-21:
why do it in code if it can be already done from way points?

merbb, 2018-08-21:
It's one of points in my big project in studies. I told my master to make it like you told because it takes about 3 min and its done but he told that it have to be done in code....

IamPete1, 2018-08-21:
seems abit pointless, but should be faily straight forward i think, i suggest you setup SITL for testing

merbb, 2018-08-21:
Can you explain your thought more?

DavidIngraham, 2018-10-12:
Hello @merbb. I hope your project has gone well. I'm going to close this issue because it is not a bug or a feature request. If you need help, I'd recommend posting in the development forum at disucss.ardupilot.org.

closed 2018-10-12 by DavidIngraham