ArduPilot/ardupilot #9245: ChibiOS: log download speed just around 100kb/s

liang-tang, 2018-08-22:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
download 17293312 bytes log take 162s

**Version**
master branch , commit id :519be92
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Pixhawk

**Logs**
_Please provide a link to any relevant logs that show the issue_

liang-tang, 2018-08-22:
Nuttx test result:
    download 9973760 bytes take 55s

peterbarker, 2018-08-27:
Yes, this has indeed been killed recently.

44fd85ff2965948b79a540ec626db5bdc57de0a1 Downloading log6.bin - 1224704/0 bytes 53.8 kbyte/s
44fd85ff2965948b79a540ec626db5bdc57de0a1^ Downloading log6.bin - 831488/0 bytes 269.8 kbyte/s (0 retries 0 missing)

liang-tang, 2018-08-28:
https://github.com/ArduPilot/ardupilot/commit/7897fbc3c16f359c0f9859eb5994cd7b7d5bf2c9  alive again.

closed 2018-09-03 by liang-tang