ArduPilot/ardupilot #9248: ChibiOS: can't parse invisible param.

liang-tang, 2018-08-22:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
[image]

_Please describe the problem_
can't parse invisible param.
**Version**
master branch , commit id :519be92

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

liang-tang, 2018-08-24:
when switch back to nuttx（Plane 3.8.4），same issue. but when reset all params to default using MP，this issue can be solved，this is a param incompatible issue between chibios and nuttx？

peterbarker, 2020-11-25:
I think this was a parsing problem, and I vaguely recall it being fixes.

closed 2020-11-25 by peterbarker