ArduPilot/ardupilot #9251: Rover refuse arming and wait until EKF is ready for non manual modes

khancyr, 2018-08-22:
### Bug report

**Issue details**

Currently rover allow arming directly on guided for example even if EKF isn't ready as it use DCM as fallback.
We should rework that arming check to specify that we want the EKF not the DCM.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

rmackay9, 2018-08-22:
I would be happy insisting that Rover always uses the EKF just like Copter does.  There is one situation where DCM is useful, in that if there's a loss of GPS, DCM degrades more slowly than the EKF.  Still, this is not an issue that we've ever seen on a production vehicle (as far as I know).  What I have seen on production vehicles (and in the simulator) is crazy drive-aways when using DCM.  This is perhaps because the velocity estimates are very bad.

rmackay9, 2019-02-20:
This is done now in Rover-3.5 which includes the EKF check.

closed 2019-02-20 by rmackay9