ArduPilot/ardupilot #9267: Feature Request: Support Frsky Smart Port Telemetry on Matek F405-WING, Omnibus F4 Pro Like FC

davidngrc, 2018-08-23:
I am using OpenTx 2.2.2 in my X9D Plus + R9 Mini SBUS Firmware

On Pixhawk, we could use Frsky Smart Port Telemetry by adding extra hardware like "FrSky Telemetry Cables" or "Teensy Board"
mention here http://ardupilot.org/copter/docs/common-frsky-telemetry.html

But the Matek F405 WING, Omnibus F4 Pro now supported by ArduPilot.
Since Matek F405 WING, Omnibus F4 Pro like FC are support SPort Telemetry without extra hardware in iNav, CleanFlight, BetaFlight.
So I was hopping the latest firmware 2018-08-23 http://firmware.ardupilot.org/Plane/beta/MatekF405-Wing/   
will work, but it does not.

because in here, it said the TX2 (SPort) pin is disabled.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/MatekF405-Wing/hwdef.dat

I try T1, T2, T3, T5, 
with serial protocol set to 4=frsky sport or 10=frsky sport passthrough
same not working, no new sensor.

I even try get the telemetry from this latest lua script
https://github.com/jplopezll/OpenTX_FrSkySPort_passthrough_master
same not working, I guest this script need a TTL-Serial inverter adaptor as mention in the readme.

utkinpol, 2018-08-26:
That was discussed on chibios thread and would require alteration on current uart driver to support 2 new options similar to betaflight to enforce inversion of uart output on f4/f7 controllers and also needs to add half-duplex protocol support.

davidngrc, 2018-09-14:
@utkinpol  could you add a link to that discussion? thank you.

Ant73, 2018-10-07:
I would also like to know more about this.

Ind0man, 2018-11-01:
Hi, 
Looking at:
[https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers](https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers)
we can see Rx and Tx pads of X4R and XSR are exposed, so shouldn't it work straight out of the box soldering to those pads?
It avoids using any diy cables and gives us a real uart port to deal with without having to change anything in the protocol.
Am I missing something ?

ThomasH3, 2018-11-03:
I have telemetry working on Omnibus F4 pro on UART 1 - frsky XSR, on Matek F405 I cannot get telemetry

Ind0man, 2018-11-03:
After testing soldering to those pads most of the sensors are detected but not updated. So basically not working. And I am with a Matek F405 CTR as well.

ThomasH3, 2018-11-03:
can you try …. 1N4007 between RX5 - TX5 (TX on cathode, RX on anode to uninverted sport of XSR) ……. select serial 4 ...  baud 57,  protocol 10

utkinpol, 2018-11-03:
https://gitter.im/ArduPilot/ChibiOS

Discussion is here. I had same bad results trying to use noninverted input. Code needs new features added to support same uart control betaflight has.

Ind0man, 2018-11-03:
Do you mean bridging rx and tx on Fc side? What's the point of doing so?

utkinpol, 2018-11-03:
Bridge with a diode to make it talk half dulex, sort of. A bad solution, but works. Proper solution is to configure fc uart into halfduplex mode

Ind0man, 2018-11-03:
TX on anode and rx on cathode did the trick indeed.

Ind0man, 2018-11-12:
Referring to Oscar Liang's website, solder pin A to tx on Fc and pin B to rx of the same uart on Fc.

 On Fc side bridge tx and rx with anode on TX and cathode on rx. 

Check hwdef of your own board, you might find something as following:

# Now the UART order. These map to the hal.uartA to hal.uartF
# objects. If you use a shorter list then HAL_Empty::UARTDriver
# objects are substituted for later UARTs, or you can leave a gap by
# listing one or more of the uarts as EMPTY.

# The normal usage of this ordering is:
# 1) SERIAL0: console (primary mavlink, usually USB)
# 2) SERIAL3: primary GPS
# 3) SERIAL1: telem1
# 4) SERIAL2: telem2
# 5) SERIAL4: GPS2
# 6) SERIAL5: extra UART (usually RTOS debug console)

# order of UARTs (and USB)
UART_ORDER OTG1 USART6 USART1 UART4

In your case it seems telem1 is uart 1 and serial1_protocol should indeed be set to 10.

dazex, 2018-11-12:
Thanks so much for writing all that @Ind0man 

It worked exactly as you described. I missed the fact that there was an A and B point for S.PORT. 

I wonder if the R-XSR has a "B" point we can tap into somewhere. 

Anyways, happy that this is working. Set SERIAL1_PROTOCOL to 10 and rebooted the FC. And all additional sensors were discovered.

utkinpol, 2018-11-12:
> Thanks so much for writing all that @Ind0man
>
> It worked exactly as you described. I missed the fact that there was an A and B point for S.PORT.
>
> I wonder if the R-XSR has a "B" point we can tap into [...]

https://github.com/ArduPilot/ardupilot/pull/9749

you no longer need to use a hack - this PR will bring inversion control as an option in arducopter.

dazex, 2018-11-12:
@Ind0man 
I wanted to use use the connector on the XSR...so I removed the inverter to free up the S.PORT pin. And then removed the resistor circled in the yellow box to remove free up the CPPM pin.

Then decided to just do the diode bridge between points A and B on the XSR itself.

Here's what it looked like before I Liquid Electrical Tape things down and then heat shrink my XSR RX.

[http://www.hacksmods.com/wp-content/uploads/2018/11/xsr_frsky_ardupilot_telemetry_mod.jpg](url) 

So now, the modded XSR can use the original plug. And the CCPM and SPORT position gets soldered to the Uart's RX and TX pads respectively on the FC.

Ind0man, 2018-11-12:
That looks really nice. Thanks for sharing ^^

davidngrc, 2018-12-05:
I would really love to test it, but there is not beta in
http://firmware.ardupilot.org/Plane/beta/MatekF405-Wing/

http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html#building-setup-windows-cygwin
I try to compile my own build in windows 2012, but fail at 
pip2 install pymavlink

```

    copying ./dialects/v20/uAvionix.xml -> build/lib.cygwin-2.9.0-x86_64-2.7/pymavlink/dialects/v20
    running build_ext
    building 'mavnative' extension
    creating build/temp.cygwin-2.9.0-x86_64-2.7
    creating build/temp.cygwin-2.9.0-x86_64-2.7/mavnative
    gcc -fno-strict-aliasing -ggdb -O2 -pipe -Wimplicit-function-declaration -fdebug-prefix-map=/usr/src/ports/python2/python2-2.7.14-1.x86_64/build=/usr/src/debug/python2-2.7.14-1 -fdebug-prefix-map=/usr/src/ports/python2/python2-2.7.14-1.x86_64/src/Python-2.7.14=/usr/src/debug/python2-2.7.14-1 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Igenerator/C/include_v1.0 -Igenerator/C/include_v2.0 -Imavnative -I/usr/include/python2.7 -c mavnative/mavnative.c -o build/temp.cygwin-2.9.0-x86_64-2.7/mavnative/mavnative.o
    error: command 'gcc' failed with exit status 127

    ----------------------------------------
Command "/usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-fXgWVL/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-s4zqnv/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-fXgWVL/pymavlink/

```

davidngrc, 2018-12-07:
@utkinpol @tridge

just download  2018-12-07 ArduPlane V3.10.0-dev
http://firmware.ardupilot.org/Plane/latest/MatekF405-Wing/

and 2018-11-25  v1.3.61
http://firmware.ardupilot.org/Tools/MissionPlanner/MissionPlanner-latest.zip

Test it on Matek F405-WING     for UART 1, 3, 5     ( FrSky S8R + X9DP )
SERIALn_BAUD = 57  which is 57600
SERIALn_OPTIONS = x  where x I try set to 1, then 2, then 3, then 7 (saved setting, power off and on each time)
SERIALn_PROTOCOL = 10  which is FrySky SPort Passthrough

I also try connect the S8R SPort pin to a MatekF405-Wing TX pin, found nothing, then connect to MatekF405-WingRX pin, and redo the test.
all no new sensor.

shellixyz, 2019-01-17:
Hello,

I've a got a R9 slim and a Kakute F7 and flashed ea2970bf2 (master's HEAD at the time of this post). I'm trying to get S.Port working.

S.Port from the R9 slim is connected to the the TX pad of USART2 (SERIAL2) and I set `SERIAL2_PROTOCOL` to 4 and `SERIAL2_OPTIONS` to 7 but I don't get any sensors appearing when enabling "Discover new sensors".

Is it the way it should be wired ?

auturgy, 2019-01-17:
@shellixyz the thread here might help: https://discuss.ardupilot.org/t/kakute-f7-aio-s-port-telemetry/34106/60

shellixyz, 2019-01-17:
@auturgy Ok thank you. Looks like I'm not the only one that can't make the Slim work with Ardupilot.

choyado, 2019-02-13:
Update to R9 slim today. I will test tonight
2019-02-13 | 190201 | Fix the incorrect telemetry issue
-- | -- | --

choyado, 2019-02-14:
I can confirm that it works on an F405-wing with an R9-slim + and the latest firmware. I used the settings below:

SERIAL4_BAUD 57
SERIAL4_OPTIONS 4
SERIAL4_PROTOCOL 10

maciek01, 2019-02-14:
any chance this could be addressed for arducopter/matek405wing/X8R - currently no telemetry

shellixyz, 2019-02-14:
What about SERIALx_PROTOCOL 4 ? To me it doesn't seem to send the right sensor IDs.

choyado, 2019-02-14:
> any chance this could be addressed for arducopter/matek405wing/X8R - currently no telemetry
The R9slim + has an inverted(actually uninverted) s.port pin like the the R9mm. I think the issue with the X8R is that it only has standard s.port. You can probably hack the receiver to find the uninverted signal.

choyado, 2019-02-14:
> What about SERIALx_PROTOCOL 4 ? To me it doesn't seem to send the right sensor IDs.

I can try protocol 4 tonight. However, protocol 10 pass-through telemetry with Yaapu lua scripts is pretty impressive.  Try it.

shellixyz, 2019-02-14:
Indeed I should try it :+1:. Please test protocol 4 anyway if you can.

MATEKSYS, 2019-02-15:
Tested with latest firmware
http://www.mateksys.com/?portfolio=f405-wing#tab-id-6

non-inverted (hacked) S.Port signal
TX1 pad of F405-WING
SERIAL1_BAUD   57
SERIAL1_OPTIONS  7
SERIAL1_PROTOCOL  4

or TX6 pad of F405-WING
SERIAL6_BAUD   57
SERIAL6_OPTIONS  7
SERIAL6_PROTOCOL  4

choyado, 2019-02-15:
I could not get protocol 4 working with the settings suggested by Mateksys. I tried it on both TX1 and TX6 connected with the RX1 (un-inverted) pin on the R9 slim +. Again, Protocol 10 works fine with Yaapu telelmetry scripts.

ThomasH3, 2019-02-15:
Hi Mateksys .... could you send me the pins for F405-CTR ...... I was using a hacked XSR on TX5 - RX5. That was working well. ... But I cannot get telemetry with an R9 mm (uninverted SPORT) on TX5.

maciek01, 2019-02-15:
> Tested with latest firmware
> http://www.mateksys.com/?portfolio=f405-wing#tab-id-6
> 
> non-inverted (hacked) S.Port signal
> TX1 pad of F405-WING
> SERIAL1_BAUD 57
> SERIAL1_OPTIONS 7
> SERIAL1_PROTOCOL 4
> 
> or TX6 pad of F405-WING
> SERIAL6_BAUD 57
> SERIAL6_OPTIONS 7
> SERIAL6_PROTOCOL 4

@MATEKSYS  can you clarify that SERIAL6 above was tested with regular inverted s.port signal out of X8R or similar.

MATEKSYS, 2019-02-16:
@maciek01  S.Port signal doesn't work,   need non-inverted S.Port.

@ThomasH3  I have no R9mm, you might try “Inverted S.Port” pad on R9mm,  I don't understand why Frsky mark it "inverted" :-)

ThomasH3, 2019-02-16:
I had to swap UART's, following setup is working (R9 MM to Matek F405-CTR , STD)
Serial1: Telemetry - TX3 - SPORT (Inverted)…. Baud 57, Option 4, Protocol 10
or
Serial2: Telemetry - TX4 - SPORT (Inverted) …. Baud 57, Option 4, Protocol 10
Serial3: GPS - RX/TX 1 ….. Baud 57, Option 0, Protocol 5

mazzaluka, 2019-02-21:
Hello , I'm Gian luca. I'm tryng exactly what you last wrote on MATEK WING FRSKY Telemetry capability. With latest AP FW Stable and FRSKY SLIM+ RX1 to TX1 or TX6  matek pin. On my mission planner(3.9.5)I've tested both SERIALx_PROTOCOL 4 or 10 and SERIALx_BAUD 57, but I'Can't set any SERIALx_OPTION by the fact I'v not this parameter in the list. In any case I even tried to DISCOVER on QX7 new SENSORS but can only see RSSI and VOLTAGE. AM I missing something? TK for any kind of suggestion

PS A doubt: could I discover new sensors, in any case, even if did not install LUA yaapu yet? In my opinion should be so

mazzaluka, 2019-02-21:
HELLO. 
AS MATEKSYS and CHOYADO report, seem possible to get telemetry both on MATEK WING TX1 and TX6 on RX1 FRSKY SLIM+ RX. I follow all procedure correct, trying any combination but cannot get any discovered new sensor on my QX7 Taranis. I didn't yet install the LUA yaapu script, but , i souppose, should be different and could be able to discover sesnsors anyway, does it?
Another difference i' got, I can't set the SERIALx_OPTION parameter on MPby the fact I've not it in the full par list. My MP ver is the V3.9.5 and FW is the latest stable kept on official AP repository. Can any one suggest any my possible missing action?
TK  a lot

choyado, 2019-02-21:
A couple things. The serial options parameter is only available on 3.10 (latest) version at this time. If you are using passthrough telemetry (protocol 10) you cannot discover sensors until the script is running. Lastly I was not able to get protocol 4 working on my FC.

mazzaluka, 2019-02-21:
Great, i will try on tomorrow. But how is possible options 7 if on the official AP Full Parameter List, it's only till 4?

mazzaluka, 2019-02-21:
EX:
[image]

choyado, 2019-02-21:
I think 7 is options 4,2, and 1 at the same time.

mazzaluka, 2019-02-21:
Ok, Tk Choyado, I'll try on tomorrow. A curiosity: on which TX you're using it?

mazzaluka, 2019-02-22:
Confirm. Work great according to choyado guide. Tk a lot

choyado, 2019-02-22:
> Ok, Tk Choyado, I'll try on tomorrow. A curiosity: on which TX you're using it?

I am using a Taranis TX.

ilya-88, 2019-02-26:
I use a single wire connection from tx to an inverted s.port to r9mm. Now there is only GPS data. If you change the destination and parameters of the port, then after turning on the flight controller, the data will appear for 1 second (the correct data and all sensors), but then disappear

mazzaluka, 2019-02-27:
I don't know the R9mm function, even if it should be same of Slim+. Have you update latest rx fw on frsky? F405-Ctr really don't know, mine is F405-Wing

ilya-88, 2019-02-27:
Yes, I updated it, and the equipment and the receiver and transmitter, the data appear during the controller loading and immediately disappear

choyado, 2019-02-27:
Are you using passthrough or frsky x telemetry? I could not get x to work.

ilya-88, 2019-02-27:
Options 4, protocol 10, I only have gps displayed, there are no others ...

choyado, 2019-02-27:
Are you using a lua script like Yappu? Protocaol 10 "(Passthrough) telemetry will not populate the the telemetry sensors on the Taranis unless a lua script is running.

ilya-88, 2019-02-27:
Yes you are right !!! in menu number 12 on taranis x7

ilya-88, 2019-02-27:
earned !!! It turned out you need to install the script yappu, only with coordinates it is not clear gps, only 4 digits displays

ilya-88, 2019-02-28:
And why in this firmware 7 and 8 channel are not available? And others are the same, there is no autotune. 3.7 firmware is the latest

dnick971, 2019-03-04:
Matek F405 wing , arduplane 3.10-dev, X8R SPORT connected to SERIAL1 i can't get telemetry to work with my frsky radio QX7S with opentx 2.2.3 (yapuu or flightdeck don't receive data). Does the SERIALX_OPTION work with F4 board ?

choyado, 2019-03-04:
Did you try serial option 4?

dnick971, 2019-03-04:
Yes but i got a "no telemetry" error message with yapuu, flight deck stay in its "waiting for autopilot" state. 
The first forum post deals with a kakute F7, but the F405 wing has a STM F4 is this the problem ?
Is it necessary to "hack" the craft and theory cable ?

choyado, 2019-03-04:
I have an f405 wing also. I am connecting directly to the FC (no craft and theory cable). However, I am using a R9slim plus and the uninverted (inverted according to FRsky) s-port pin. F4 boards can't handle inverted telemetry. There is a way to get the uninverted signal off of a X4R....not sure how to do it on an X8R

rafaroldan, 2019-03-12:
> I have an f405 wing also. I am connecting directly to the FC (no craft and theory cable). However, I am using a R9slim plus and the uninverted (inverted according to FRsky) s-port pin. F4 boards [...]

Hi! Could you please post the parameters for SERIAL1 you are using? And please, confirm that only a single cable from TX1 (on the F405WING) to the receiver S.PORT pin (on the R9slim) is needed. I'cant get it working.....
Thanks!

choyado, 2019-03-12:
Do you have the slim plus? 

SERIAL4_BAUD 57
SERIAL4_OPTIONS 4
SERIAL4_PROTOCOL 10

rafaroldan, 2019-03-12:
Thanks! Yes the R9 Slim+

choyado, 2019-03-12:
Make sure to use the s.port pin that is labeled inverted.

[22 more replies]

closed 2019-06-01 by OXINARF