ArduPilot/ardupilot #9268: Chibios not respecting BRD_SAFETY_MASK

Naterater, 2018-08-23:
### Bug report

**Issue details**

Plane 3.8.5 nuttx has my servos on during boot. Same parameters on Chibios doesn't work the same way.

**Version**
Plane 3.9 chibios and up

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Cube Black

**Logs**
https://www.dropbox.com/s/met51kinrish0ru/00000039.BIN?dl=0

Pedals2Paddles, 2018-08-23:
I just opened a similar one this morning too.  https://github.com/ArduPilot/ardupilot/issues/9266

Naterater, 2018-08-25:
Just experienced a crash due to a reboot mid-flight. I had 20 seconds with no control in manual mode since the servos were stuck at their last position.

Reboot likely caused by another electrical issue, but nonetheless I should have been able to recover in manual.

WickedShell, 2018-08-25:
@Naterater ChibiOS does not provide the mixer to the IO processor. This is a known limitation, what this means is that you won't have any control of the vehicle except when the FMU is booted. If you want to be able to use the IO processor for this you need to be running PX4 builds.

A couple of side notes: Modes have no effect on IO processor passthrough, if the FMU stops talking the IO processor immediately falls over to manual if it has a valid mixer loaded. Depending on what caused your reboot (IE power failure) you wouldn't have had manual control anyways.

Naterater, 2018-08-25:
My point is that the FMU booted again and I should have been able to have control. The FMU was booted during the crash. I was only unable to actuate servos because of the safety switch. I even had telemetry momentarily after the reboot. Not enough time to actuate the safety switch via telemetry though. I think that RC should be passed through ASAP on boot if the RX is sending manual mode. This gives at least some potential for recovery without going back to PX4 builds.

Naterater, 2018-09-17:
Once again I experienced a reboot mid-flight causing another crash. This is becoming quite frustrating.

Before reboot: https://www.dropbox.com/s/nb7z07vmptoc7nh/00000007.BIN?dl=0
After: https://www.dropbox.com/s/ureppt412vefvzc/00000008.BIN?dl=0

Naterater, 2018-10-04:
@tridge, have you had a chance to get this mask working again?

tridge, 2018-11-06:
this is fixed in master now, please re-test and confirm

closed 2018-11-06 by tridge