ArduPilot/ardupilot #9282: Build fails at make_intel_hex.py

auturgy, 2018-08-25:
### Bug report

**Issue details**
My system (Ubuntu 18.04) is defaulting to Python3 (I think by default it would be python2, but I'm doing some other things with python3).  At completion of compilation, the following error is thrown:

```
 Traceback (most recent call last):
  File "/home/james/Development/ardupilot/Tools/scripts/make_intel_hex.py", line 4, in <module>
    import intelhex
ModuleNotFoundError: No module named 'intelhex'

Waf: Leaving directory `/home/james/Development/ardupilot/build/fmuv3'
Build failed
 -> task in 'bin/arducopter' failed (exit status 1): 
	{task 140125291077832: build_intel_hex arducopter.bin,fmuv3_bl.bin -> arducopter.hex}
 (run with -v to display more information)
```

_Please describe the problem_
Python2 package intelhex is not found by make_intel_hex.py although it is present on the system.
Fixed by https://github.com/ArduPilot/ardupilot/pull/9279

**Version**
Not version dependant (build system)
**Platform**
[  X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Build system

**Hardware type**
Build system

**Logs**
Console text pasted above

WickedShell, 2018-08-26:
@auturgy intelhex works with python3, just pip install it. Python3 is broken for a number of other reasons at the moment in the uavcan module, but I was building with intelhex no problem before that on a python3 system. Is your above failure during the configure step?

auturgy, 2018-08-26:
> @auturgy [1] intelhex works with python3, just pip install it. Python3
> thread [3].
> Links:

I did pip install it....(pip, pip2, pip3)
waf configures fine.  It builds all the way through until the generation 
of the hex from the bin.
The cause is miniconda3, which during install appends the .bashrc and 
messes with python globally.
I worked around it in a different way on my own machine using sym links, 
because it messed with ROS too, but the PR I made also works (for this 
particular issue).  Most people don't want to have to add random sym 
links to work around config issues.

WickedShell, 2018-08-26:
The main problem with the PR is that python2 may not exsist and then people still have to do symlinks. Given that it does work under python3 I think that PR is probably not the best path forwards...

What we should probably do is instead change the task to pass the python runtime in. So that would be to change: https://github.com/ArduPilot/ardupilot/blob/a4e50e1626b9de894f8f32a14fc03a30f220a858/Tools/ardupilotwaf/chibios.py#L116 to look like more like https://github.com/ArduPilot/ardupilot/blob/a4e50e1626b9de894f8f32a14fc03a30f220a858/Tools/ardupilotwaf/chibios.py#L96 (note the `${PYTHON}` tag). That should allow you to do:
```
python2 waf configure --board=fmuv3
python2 waf plane
```

The main advantage of this is that it allows you to specify the python environment you want without forcing people with working python3 installs onto python2 (which may not be working for them). It's a bit more verbose, but it's a lot easier then symlinks, and is currently added to work around some other (newest stable) Python3 things anyways.

OXINARF, 2018-09-26:
The change proposed by @WickedShell has been merged, so you can know force the Python version that waf is run with (and so intelhex).

OXINARF, 2018-09-26:
Thanks to @WickedShell for making me notice I looked at the wrong line of code, this isn't in master after all, re-opening.

peterbarker, 2020-11-25:
This all works now.

closed 2020-11-25 by peterbarker