ArduPilot/ardupilot #9295: AIS Support

grcasanova, 2018-08-27:
# Support AIS collision avoidance

It would be good to implement a rudimentary vessel detection and avoidance for ArduBoat using an AIS receiver.
AIS - Automatic Identification System - is a transponder-based system mandatory on voyaging ships with 300+ gross tonnage streaming data about speed, location, heading and so on.

## The Algorithm
### Basic
A simple way of dealing with AIS would be to create a 2D map of all vessels nearby and update their position keeping track of their ID.

### Advanced
Using heading and speed it is possible to define a speed vector for each identified vessel and create a more reliable map, replanning the boat path in order to avoid other ships' routes.

## The Sensor
### CAN-based receivers
Most AIS receivers have a GPS antenna inside and output NMEA messages via Can Bus protocol.

### RTL-SDR
My favourite approach, simply using a RTL USB dongle it is possible to interpret AIS messages. It requires having a companion computer.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

**Discussion**
[https://discuss.ardupilot.org/t/ais-basic-collision-avoidance/30461](https://discuss.ardupilot.org/t/ais-basic-collision-avoidance/30461)

peterbarker, 2018-08-27:
On Mon, 27 Aug 2018, Giammarco R. Casanova wrote:

>                                                              Support AIS collision avoidance

You might check out what tridge is cooking up for the Medical Express 
Challenge 2018.  It uses ArduPilot's AP_Avoidance to try to continue 
towards a waypoint while maintaining minimum distances from obstacles both 
stationary and dynamic.

grcasanova, 2018-08-27:
I would avoid commercial AIS receivers because they are power hungry and heavy.
dAISy and SDR support should be fine for now as are the two cheapest (and small-sized) solutions.

@peterbarker
Seems a good solution, what types of data does this algorithm need in order to work? Does it keep track of obstacles using IDs or regen a new map for every dataset coming in?

peterbarker, 2018-08-27:
On Mon, 27 Aug 2018, Giammarco R. Casanova wrote:

> @peterbarker
> Seems a good solution, what types of data does this algorithm need in order to work? Does it keep track of obstacles using IDs or regen a new map for
> every dataset coming in?

AP_Avoidance can be fed via mavlink ADSB messages or 
mavlink-vehicle-position data.  tridge has a proxy which is converting 
asterix data into adsb data as it was a relatively easy way forward.

obstacles are id'd.

grcasanova, 2018-08-27:
We could try using MAVlink vehicle position if that supports heading and speed values in order to calculate the vessel path

peterbarker, 2018-08-27:
On Mon, 27 Aug 2018, Giammarco R. Casanova wrote:

> We could try using MAVlink vehicle position if that supports heading and speed values in order to calculate the vessel path

Projection is done.

Depending on the protocol information, it might be better or easier to 
creae a new mavlink message type specifically to carry the AIS data - this 
is the way it was done for ADSB:

https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4194

rmackay9, 2018-08-28:
I think it would be best to start with a driver to read in the info from an AIS device.  Preferrably an easy DIY device using an RPI or similar..

grcasanova, 2018-08-28:
I am using AISDeco2 (http://xdeco.org) with a SDR dongle and it works pretty well, considering I am still building a proper 162MHz vertically polarised antenna. It runs on Windows and Linux and supports ARM (Raspberry Pi).

To prioritise things, do you believe it is important to have MAVlink messages for AIS? Should we wait for _tridge_ implementation of the obstacle avoidance system or start to experiment with already-implemented avoidance systems?

peterbarker, 2018-08-29:
On Tue, 28 Aug 2018, Giammarco R. Casanova wrote:

> To prioritise things, do you believe it is important to have MAVlink
> messages for AIS? Should we wait for tridge implementation of the obstacle

Not really; if you're going via a Pi you could translate from AIS to ADSB 
- I doubt you'll find anything in AIS that wouldn't fit in an ADSB packet.

> avoidance system or start to experiment with already-implemented avoidance
> systems?

The current on-board avoidance code is limited to things like "run away" 
or "hold in place" and the like.  Given there's no time-frame for the 
auto-mode avoidance stuff tridge is working on coming into mainline, you 
might consider investigating off-board systems for avoidance.

auturgy, 2018-08-29:
Reusing the adsb message would be ok for testing but shouldn’t be recommended for actual use without warnings. It’s a hack, and could potentially/accidentally be rebroadcast by a gcs into the inter web.
Either a generalised transponder message with an enum for transponder type, or a new message.
Just my 2c.

Regards,

James

rmackay9, 2018-08-30:
I think re-using much of Copter's Avoid_ADSB flight mode is the best way forward.  Writing a new driver to pull the data from the AIS sensor makes sense.  I don't think that an offboard solution is the best way forward because it will be difficult to integrate well with other existing failsafes.

grcasanova, 2018-08-30:
What kind of AIS receivers should ArduPilot support?

As long as the ArduPilot AIS code accepts external input from a companion computer the offboard solution will be easily expandable for those interested, as I was doing with ADSB and a SDR dongle.

rmackay9, 2021-08-10:
AIS support is included in Rover-4.2.0 and we have a wiki page here: https://ardupilot.org/rover/docs/common-ais.html

closed 2021-08-10 by rmackay9