ArduPilot/ardupilot #9303: Rover: fix non-GPS wheel encoder feature

rmackay9, 2018-08-28:
With the recent introduction of the balance bot code we increased the rate that the wheel encoder info is sent from the main vehicle code to the EKF.  It was 20hz but it's been increased to 50hz.

It seems that currently our EKF can only accept wheel odometry data at up to 20hz.  The check is done [here in the NavEKF3_core::writeWheelOdom() method](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_Measurements.cpp#L142).

There are two possible solutions:

- reduce the [sensorIntervalMin_ms definition in AP_NavEKF3.h](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3.h#L448) to 20 (or 10?).  Note that this will increase the RAM usage of the EKF
- modify [Rover::update_wheel_encoder()](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/sensors.cpp#L69) so that it calls the wheel_encoder.update() function at a high rate (like 50hz) but then consolidates the changes and sends to the EKF (using the EKF3.writeWheelOdom() call) at a lower rate.

robotiste, 2019-03-14:
@rmackay9 do you have any rosbag of this?

rmackay9, 2019-03-15:
@robotiste, I do not I'm afraid .. but this issue doesn't directly involve ROS I think.. the wheel encoder hardware is usually connected directly to the flight controller and as far as I know we don't accept wheel encoder inputs via MAVLink..  I guess we do send wheel-encoder messages to ROS/mavros now..

rmackay9, 2019-10-22:
This is fixed in master so closing.  Hopefully this will be tested as part of the Rover-4.0.0 beta..

closed 2019-10-22 by rmackay9