ArduPilot/ardupilot #9312: Critical: Throttle can run while disarmed and safe at reverse thrust setups on reboot

palm369, 2018-08-29:
### Bug report

When you reboot the plane via mavlink command the motor might run full power backwards when you have set up reverse thrust.
Might only happen If motor ran before (so it is initialized)

Exact setup:
Turnigy 60A ESC with BLheli on it set up to Bidirectional 1000 - 2000 us with neutral at 1500 us

Throttle servo channel set up to the same.
SERVO3_FUNCTION,70
SERVO3_MAX,2000
SERVO3_MIN,1000
SERVO3_REVERSED,0
SERVO3_TRIM,1500

RC3_DZ,0
RC3_MAX,2004
RC3_MIN,900
RC3_REVERSED,0
RC3_TRIM,980

**Version**
AP 3.9.1 B3 and earlier, but only the NuttX version, does not happen with ChibiOS

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Only tested on PH2.1

Naterater, 2018-11-08:
@tridge, I'll see if I can test this soon with the new IO.

IamPete1, 2021-07-26:
@palm369 states that this is not a issue on ChibiOS so I'm going to close.

closed 2021-07-26 by IamPete1