ArduPilot/ardupilot #9315: Class for a unique drone design

KisorAlaguvel, 2018-08-29:
Please someone help add a new class for conversion of commands into motor outputs based on a new mathematical expression specific for my own unique drone design..

KisorAlaguvel, 2018-09-05:
Hello DavidIngraham,
I've written about the isse in detail with images attached here : https://engineering.stackexchange.com/q/23556/17365

KisorAlaguvel, 2018-09-05:
Hi rmackay9,
I believe it is not the duplicate of the issue linked. Please visit the link above given in another answer.

KisorAlaguvel, 2018-09-05:
Supposing if I add code for my new design successfully, now do I need to change something in the dronekit python as well provided I'm using it?
What's the case for the same in dronekit SITL? 
As I understand so far, Dronekit is just for communication to ArduPilot over MAVLink in python..right?

IamPete1, 2021-07-26:
We now support custom motor layouts via scripting.

closed 2021-07-26 by IamPete1