ArduPilot/ardupilot #9316: Onboard Compascal stuck if in certain configuration.

AndKe, 2018-08-29:
### Bug report

**Issue details**
This config:

COMPASS_AUTO_ROT,1
COMPASS_CAL_FIT,32
COMPASS_DEC,0.15431185
COMPASS_DEV_ID,466441
COMPASS_DEV_ID2,466433
COMPASS_DEV_ID3,131874
COMPASS_EXTERN2,1
COMPASS_EXTERN3,0
COMPASS_EXTERNAL,1
COMPASS_OFFS_MAX,850
COMPASS_ORIENT,0
COMPASS_ORIENT2,29
COMPASS_ORIENT3,0
COMPASS_PRIMARY,0
COMPASS_USE,1
COMPASS_USE2,0
COMPASS_USE3,0

This config will perform internal calibration on device 466441 just fine it will show progress along the way in QGC or mavproxy magical_ui

device 466433 is the best-mounted magnetometer.
so I do change parameters like that:
COMPASS_PRIMARY,1
COMPASS_USE,0
COMPASS_USE2,1

now, I cannot get any progress during calibration, and it will never complete.
However; doing an older calibration (by GCS)  , does produce valid calibration for both first and second device (proving that both are providing data)

If I enable first and second device, then the progress will stop once first is calibrated.
---------------------------
Another clue: I have another drone with the same 466433 device, but in first place: 
COMPASS_DEV_ID,466433
COMPASS_DEV_ID2,131874
COMPASS_DEV_ID3,263178
 And this setup has no problem calibrating it using internal calibration.  
 
**Version**
3.6.0RC8 (and 3.5.5 too)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octa-quad

**Hardware type**
Pixhawk 2.1

**Logs**
n/a

rmackay9, 2018-08-29:
Great, thanks for the detailed report!

IamPete1, 2021-07-26:
I think this was just a issue that not using the first compass confused AP, lots of changes to how that stuff works now, so I'm going to close.

closed 2021-07-26 by IamPete1