ArduPilot/ardupilot #9326: Chibios, Revo mini // Feature Request: one more Uart

MFlghts, 2018-08-30:
**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Please add one more Uart to the free PWM, or better RC-in pins to be used for frsky passthrough.
Assuming that Flexi port is set for i2c.

IamPete1, 2018-08-31:
see: https://github.com/ArduPilot/ardupilot/pull/8984

I have gps, passthrough telemetry and wifi telemetry running on mine, its a fairly straight forward edit of the hwdef.dat and recompiling. I do think the 3 UARTS and rc in on the 2nd rc in pin should be the default.

MFlghts, 2018-09-05:
Please add changes of the hwdef.dat here for the third Uart.

IamPete1, 2018-09-05:
A hwdef.dat for three UARTS is in the linked pull request

IamPete1, 2021-07-26:
Looks like we now have three serial ports, and you can have four via alt-config. 

https://ardupilot.org/copter/docs/common-openpilot-revo-mini.html

In any-case I'm not sure you can even buy this board anymore.

closed 2021-07-26 by IamPete1

RMFPV, 2021-07-30:
I don't see any changes in docs about that.
Can you please describe what exactly is alt-config now?

Yes it's not easy to buy it, but some people still use it, as it's a small, nice and reliable fc.