ArduPilot/ardupilot #9340: BLHeli passthrough not working for Tailsitter

palm369, 2018-09-02:
### Bug report

When tailsitter (Q_frame_class = 10) is configured with servo functions of Aux 1+2 set to 73+74 it **does not** recognize the escs in the BLHeli suite.

Exact same setup but with quadplane (Q_frame_class = 1) configured , servo functions of Aux 1+2 set to 33+34 it **does** recognize the escs in the BLHeli suite.

It must be frame class and servo functions be change to work. One alone does not work.

**Version**
AP3.9.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tailsitter

**Hardware type**
PH1

**Logs**
nope

robustini, 2018-09-03:
Here work fine, have you installed "ChibiOS" (imperative)?

palm369, 2018-09-03:
Yes, ChibiOS.

fmuv2 00350027 31334702 38383835
ChibiOS: 6686ee8a
ArduPlane V3.9.1 (86e1c8f1)

I checked this issue multiple times before posting.

WickedShell, 2018-09-23:
@palm369 Have you tried setting the BLH_MASK to the actual outputs you are trying to use? If you are trying to use BLH_AUTO it will fail because the automatic motor mask is always set to 0 on a tailsitter. https://github.com/ArduPilot/ardupilot/blob/a5c59ce36efee4b5fad654e941172137598c7dfc/libraries/AP_Motors/AP_MotorsTailsitter.h#L30

palm369, 2018-09-24:
@WickedShell Not quite sure about that. Your right, maybe that's the cause!

IamPete1, 2021-02-08:
I'm going to close, feel free to re-open if this is still a issue.

closed 2021-02-08 by IamPete1