ArduPilot/ardupilot #9347: Rover/Boat: mapping mode

rmackay9, 2018-09-03:
It would be very convenient for users is we had a "map" control mode that would cause the rover or boat to cover all accessible areas of an indoor space or lake while also staying within the bounds of the polygon and circular fences.

[image]

The usefulness of this potential mode comes from [mapping a lake as described in this Blog](https://discuss.ardupilot.org/t/mapping-a-lake-with-ardupilot/32401).  The user has to be quite careful about placing the waypoints so that they are close (so that we map all of the lake) but not too close to the shore (so that the boat doesn't get stuck in weeks or hit the shore).  This can be difficult especially in cases where the GPS coordinates don't exactly match the GCS's map (due to GPS errors or inaccuracies in the map service the GCS is using).

DavidA makes this clear when they say:

> the biggest issue I have found when survey is trying to plan the route as maps is not that accurate there can be up to a 10m map shift. How always have to survey the perimeter of the reservoir with GPS before planning a mission

DavidB says something similar:

> I usually do a survey grid first then a few laps around the perimeter in steering mode using telemetry and FPV after the auto grid is complete.

Below are some thoughts on how it might work for boats which could be simpler than what is required for rovers because they typically operate in a simple environment.

- the boat could at first travel straight forward in the direction it is facing when MAP mode is engaged.
- the shoreline (or other barrier) could be detected using an (optional) downward facing sonar (new DEPTH_MIN parameter would be required) or a forward facing or 360 degree lidar detects something.  It might also just detect that it can't move forward despite motors going to full (although this could be messy).
- once the shoreline is detected the vehicle might give up on moving in that direction and instead turn back 180degrees and attempt to maintain a line X meters to the left or right of the previous line.

A more advanced algorithm somehow capable of mapping behind barriers would be nice eventually.

Avalonnw, 2018-09-07:
As an option - set a point in the middle and spiral out with pre-set distance between lines. If encounter a blocked path, keep going along the blockage until get to the required distance from previous line and continue. Something like "adaptive toolpath" in machining.

rmackay9, 2018-09-07:
@Avalonnw, that's an interesting idea, thanks!

Avalonnw, 2018-09-07:
Actually, that reminds me a bit too much of robot vacuum cleaners and SLAM. My LG HomBot has this Spiral mode which is essentially the same. Is SLAM implementable on current hardware without companion computers at all?

Naterater, 2019-01-30:
From the discussions of features for other vehicles, I think this is a prime candidate for scripting instead of a built-in feature. Just my $0.02.