ArduPilot/ardupilot #9349: Plane: Crash detected and plane disarmed on GPS loss while on Air

Raviu8, 2018-09-03:
### Bug report

**Issue details**
Hi,
I had a bad crash Today where the plane got a GPS "no fix" for few moments and it suddenly got disarmed while still on air in Auto mission.
I've reviewed telemetry logs and found that the plane sent "crash detected" when disarmed. 

**Version**
3.9.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed wing

**Hardware type**
Pixhawk1

I could reproduce the bug many times using plane sitl and sitl with x-plane;

Steps are so easily and reproducible for me every time.
**Steps to reproduce**
1. Start sitl whether normally or with xplane.
2. enable crash_detect = 1 
3. Give the plane a mission, arm, then start Auto mode.
-- I've modified the ins_ params to avoid the 3d accelerometer calibration needed not to disable arming checks.
3. Allow it to takeoff and fly for a little then go to SIM_GPS_BYTELOSS and set it to 5
4. Go back to Flight Data, it would show GPS no fix. In messages within few moments you'll get "crash detected" and the plane is throttle disarmed.

Raviu8, 2018-09-03:
I've tracked this down to line 26 in is_flying where airspeed_movement is always returned as false when GPS is lost.
My plane had no airspeed sensor, so I've set arspd_type to 0. 
But on sitl I could reproduce the same buggy behavior with arspd_type is either 0 or 2.

magicrub, 2018-09-04:
Thanks for tracking this down!

tridge, 2018-09-04:
I have a PR to fix this here: https://github.com/ArduPilot/ardupilot/pull/9358

Raviu8, 2018-09-05:
I found that the "no disarm" was related to using arming_require = 0. I think this behavior was changed recently on master? As I recall well it was disarming on crash detection with "arming_require = 0".
So when set to 1 , it disarms. I think it should always disarm in crash detection whatever the arming_require is.

tridge, 2018-09-05:
@magicrub what do you think? I'm tempted to just disable crash detection if we don't have GPS lock and also don't have working airspeed. Really at that point we have no reliable way to know if we've crashed or not.

tridge, 2018-09-05:
I've added a patch to implement the "no crash if no GPS and no airspeed" approach

Raviu8, 2018-09-06:
In simulator it seems fine now, with and without airspeed sensor.
But what about the "no disarm" with "arming_require = 0" when crash is detected?

Raviu8, 2018-09-07:
I've done two real flight tests Today using PR #9366 .. To test both crash detection and dead reckoning.
It was too windy that helped find another crash detection bug.
I've activated a GPS disable switch, also added a switch for a video transmitter which is proved to work like a GPS jammer. I've tested the later before flight and number of SATs drops from 16 to less than 7 with random movements that prevents arming.
No air speed was used, and arspd_type was 0.

Here are the results;
1. Dead reckoning works fine, I think. Much better with GPS disabled, less effective with GPS jamming activated.
Tested with Auto and RTL. RTL was astonishing for me as the plane almost flew back to home by side due to relatively powerful wind.
2. Another false crash detection occurred when the plane was flying into wind with good estimated AS as displayed in MissionPlanner, but very low GS. Then I turned on the disable-GPS switch, and **instantly** it detected a crash and disarmed displaying AS of 0 and GS was too much lower than stall speed as mentioned.
By the way, in my previous attempts to debug the crash detector issue, I've tried some different conditions like the one you added in commit a536760 and it usually crashed, maybe the GPS status is not updated quickly enough here. Also if GPS regained a good fix for few milliseconds the condition is usually bypassed too and it detected a crash.

tridge, 2018-09-07:
@Raviu8 do you have a log of the false crash detection cases you just found?

Raviu8, 2018-09-08:
@tridge I could reproduce the same buggy behavior using sitl;
1. Draw a mission that goes from East to West for a distance of 7 km at 600m rel. alt. for example.
2. Set fbw min to 15 m/s, set arspd_type = 0, crash_detect = 1
3. Set sim_wind_speed to 11.5 m/s and sim_wind_dir to 260 degrees - so as the plane is flying into wind.
4. Arm, start Auto, and allow plane to achieve required alt., It should have ground speed around 4 to 5 m/s
5. Set SIM_GPS_BYTELOSS to 5 within few seconds you'd get a crash detect and it'll disarm.

I'm using your PR #9366 and cherry-picked commit a5367607 into it.

Raviu8, 2018-09-09:
Here are SITL logs showing "crash detected" bug mentioned in latest comment;
https://mega.nz/#F!LPpAXCwI!TNeBMv_gBE8kHfkMPZLJsQ

Also folder includes a bin log for a Floating Exception Error that caused SITL crash too.

Raviu8, 2018-09-09:
@tridge I think I've found something;
I've made a small change in add a debouncer in case of GPS "no fix". Patch is attached on top of your pr.
I've gradually increased the debounce times, trying each time to reproduce the bug, actually I could reproduce it up to almost 14 times (2 seconds) so I've set it finally to 21 (almost 3 seconds) when I finally could not see a "crash detected" any more on GPS loss.

I do not know if this patch is a real solution, but I hope it points to one.

[gps_crash_debounce.patch.txt](https://github.com/ArduPilot/ardupilot/files/2364531/gps_crash_debounce.patch.txt)

amilcarlucas, 2018-09-10:
21 seams to be a magic number. Does it have to do with 5Hz GPS? What happens with a 10Hz GPS ?

Raviu8, 2018-09-10:
Actually, 21 means 4.2 seconds. As crash_detection_update is called update_is_flying_5Hz , so it's 5Hz, I thought it was 7Hz.
It's 5Hz on SITL as GPS_RATE_MS is 200 by default.

amilcarlucas, 2018-09-10:
I fly with a Septentrio AsteRX GPS at GPS_RATE_MS 100. Other people do that as well.
IMHO you should replace 21 with (4.2*1000/GPS_RATE_MS)

Raviu8, 2018-09-10:
@amilcarlucas 
This is so complicated I think as there could be more than one GPS each one has its own rate.
I would try with with rate set to 100 on SITL and see what I can  get.

Raviu8, 2018-09-10:
I've done some more testing, I think a value of 16 is fine with both 5Hz and 10Hz GPS.
So it's up to @tridge now to make decision.

Raviu8, 2018-09-10:
I've also simulated a case using xplane - with my patch at 16 bounces - where I got the plane to physically crash on ground while GPS was disabled then re-enabled it after it actually crashed, within few seconds - less than 4 seconds after enabling GPS, it detected a crash and disarmed - as required.

Also tested the same with arspd_type = 2 and then with also arspd_use = 1.
Setting arspd_use =1 allowed good crash detection without GPS.

So the idea is working fine, I think.

magicrub, 2018-09-11:
Hey guys, I'll take a look at this soon

amilcarlucas, 2018-09-11:
@WickedShell thanks for clarifying that. That info should be added to the wiki. (or to the parameter description)

Raviu8, 2018-09-17:
Any updates on this?
I've done real flight tests with my patch and it seems working.

Raviu8, 2018-10-26:
@tridge @magicrub Long time no updates?

tridge, 2018-11-28:
we believe this is all fixed now (thanks Tom!)
please re-test though

closed 2018-11-28 by tridge