ArduPilot/ardupilot #9356: The Angle of IST8310 Compass on pixhawk2

jamming, 2018-09-04:
The angle of pixhawk4 GPS(with IST8310 compass) are different against the old one(with HMC5883/5983 compass). When using the pixhawk4 GPS on a pixhawk2 flight-controllor, we have to change the angle of external compass  manually.  After selecting Yaw270, Mission Planner don't pop " compass inconsistent" message again. I think that maybe a bug.

rmackay9, 2018-09-05:
@jamming, if you are using Copter, could you try with Copter-3.6.0-rc8?  It's available through the mission planner's install firmware screen after pressing the "Beta Firmwares" link.  If this link isn't visible then please go to Config/Tuning >> Planner and set the "Layout" drop-down to "Advanced".

This is beta firmware so it hasn't completed all testing.  It seems stable but please be careful.

IamPete1, 2021-08-29:
@jamming has this been resolved?

jamming, 2021-08-30:
Not yet. I had tested the arducopter-latest firmware and the arducopter-4.0.7 stable firmware,  the angle of IST8310 was still Yaw270.

IamPete1, 2021-08-30:
@jamming Thanks for the update. To be clear is this a external compass on a GPS unit? Not the compass internal to the flight controller?

jamming, 2021-08-31:
Yes, it is a external compass on a GPS unit for pixhawk2. 
There is a  IST8310 internal to pixhawk4,pixhawk4-mini and Durandal, but there is no IST8310 internal to pixhawk2.

IamPete1, 2021-09-01:
I don't think there is anything we can do to help with the orientation of compass in a external GPS. We have auto rotation now, that should pick up the correct orientation when calibrating.

closed 2021-09-01 by IamPete1