ArduPilot/ardupilot #9374: Setting to limit max current based on AMPs

estebanavv, 2018-09-06:
### Feature request

**Is your feature request related to a problem? Please describe.**
Currently Arduplane implements BATT_WATT_MAX #1670 which is the most sensible way of doing it, but this setting is not ideal for Li-Ion batteries where the volt range is large but the limiting factor are Amps.

**Describe the solution you'd like**
Create a new setting BATT_AMP_MAX and BATT2_AMP_MAX

**Describe alternatives you've considered**
Currently in order to set the max amps I calculate the watts at 2.8 volts per cell X the max amps for the battery but sometimes this allows the plane to fly but it sets a very low limit when that affect makes the plane anemic even when the battery is fully charged.

Another option is to use a blheli_32 ESC with current limiter, but I assume that this might affect EKF calculations because of the additional throttle % that won't provide more thrust.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

Naterater, 2018-11-27:
This was talked about in the DevCall recently, and the discussion was about the BATT_ library and how it would be the place for new features like this. I agree with this request.

IamPete1, 2021-08-14:
Closing in favor of https://github.com/ArduPilot/ardupilot/issues/1670

closed 2021-08-14 by IamPete1