ArduPilot/ardupilot #9376: Can soneone help me create New Code for My New Design

KisorAlaguvel, 2018-09-06:
I've written about the design, which though has some clarifications yet to be made in detail, with images attached here : https://engineering.stackexchange.com/q/23556/17365
The concept is this: Adding a third propeller spinning perpendicular to ground as a torque-reaction balancer.
[image]
[image]
Forget about the aerodynamics of air being pushed on one side that would make the drone spin unbalanced.
It can be rectified by adding another one coaxially and which spins pushing the air in opposite direction. The thing is that the other free side is now free for payload which in my case is a cutting arm. It's a drone that operates semi-autonomously in harvesting coconut, much to the aid of farmers. If anyone  is interested, please join me!
I saw all the code in ArduPilot regarding this (Ap_Motors etc.,) and I see that a lot and lot of cpp files need addition and integration with one another which is an overwhelming task for me. Can someone look into this? Is it possible to create an entirely new design in this existing setup of libraries?

KisorAlaguvel, 2018-09-06:
Please visit the link for detailed math argument presented.

auturgy, 2018-09-06:
> To me, it is just looks like Tricopter became coaxial form, checkout https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsTri.cpp
> Add two more motors accordingly

It’s actually a bit different to that: he’s looking to use torque response of the tail motor to control pitch.
That might work for very small systems, or across a very narrow performance envelope, but it’s almost certainly going to have less authority than the traditional configurations (and be wildly inefficient). I’d be interested to see a sitl model though, just out of curiosity.

Regards,

James

KisorAlaguvel, 2018-09-07:
hey @auturgy 
yeah, it is inefficient only if it is supporting with its power all the time. I intend to leave the system very little (only) to be balanced by the torque-reaction of the propeller. Just for steering purpose (when its needed) and when the payload on the other side (which is a robotic arm) extends thereby creating an imbalance-moment....
Cheers:)
**Please help me with its development..where should I start? And what are all the changes required?**
@SwiftGust hey, can you please provide me with the image of the CoaxCopter you're talking about? I cannot find in the documentation page..

SwiftGust, 2018-09-07:
@auturgy Oh I missed that, It wouldn't be a matter 
even if you use ArduCopter code without copter-heli code (if your motor torque is powerful enough...)
problem(or pain in the ass is  
how much fixed throttle should be given to the tail-rotor... (maybe a 80%?)
For the blade pitch control, it wouldn't be a problem 
just giving yaw_thrust to the tail-rotor servo instead of motor control.

@KisorAlaguvel CoaxCopter is just a coaxial version of SingleCopter(or aka Ducted-Fan)
Images, Videos are in ArduPilot documentation 
http://ardupilot.org/copter/docs/singlecopter-and-coaxcopter.html

AFAIK, first of its design is Honeywell T-Hawk (RQ-16 of US Army)
https://en.wikipedia.org/wiki/Honeywell_RQ-16_T-Hawk

although I did use that to control this kind of UAV too
https://drive.google.com/open?id=1R5pvJLZPEf_TClDJYxdovCfv8uBhPIrQ

KisorAlaguvel, 2018-09-07:
@SwiftGust So, where are the needed changes? I mean, what library files? More problematic for me is writing the new needed code in consistence with existing ones (classes, methods etc.,)
And do I need to change or add something in DroneKit Python as well?

SwiftGust, 2018-09-07:
Don't mind existing one until you get to know ArduPilot code, just change it in your local fork, mess it up as you want, original code is intact here
And the code you need to modify is what I linked earlier, Dronekit or else is nothing to consider in this stage, just focus on ArduCopter code alone.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsTri.cpp

KisorAlaguvel, 2018-09-08:
Thanks a lot:)
I'll come after messing a lot of code and adopting that to mine..!

KisorAlaguvel, 2018-09-24:
@SwiftGust 
Hey! But I see a lot of dependencies on other common documents in AP_Motors. Also, what about the math? Where are the equations of motion in the ArduCopter libraries, or is it completely depending upon feedback control?
I am asking so because the mechanism or the physics with which pitch is effected is completely new i.e. torque-response instead of thrust-response.
Please guide me in here!

lthall, 2018-09-24:
I am sorry but this is not the appropriate place for this conversation as this is a request for support on a personal project not an issue or feature request.
Please move this to the discuss forums or your own GIT repository.
Thanks.

closed 2018-09-24 by lthall

KisorAlaguvel, 2018-09-25:
Hey @lthall 
I am but sure that this is a feature request. I am intending to add a whole new frame type/drone design which I think is the most suitable for the purpose it is conceived for. It is specifically designed for the harvesting of coconuts from the coconut tree. And I will add all to the Ardupilot library once it gets the full form.

lthall, 2018-09-25:
That will be great to see @KisorAlaguvel !

The way you would do that is to communicate with other members of the community on the Discuss forums and work on the project in your personal git repository. You will get more help that way too.

When you have the code to a point where you can show the new frame type in more detail, or even in action, you can do a pull request or talk about specific features you need to finish your work.

Good luck, I hope we will see your work here soon!!

KisorAlaguvel, 2018-09-26:
Yeah. Thanks:)