ArduPilot/ardupilot #9377: Copter : How to enable read redundant rc input channels as old firmware

SwiftGust, 2018-09-06:
### Bug report
**Issue details**

I'm trying to move custom code I wrote for old firmware to ardupilot-master, 
I have a problem that
I could be able to do something switch with an additional simple code in radio.cpp in read_radio() function like read PWM from RC input 5~8 and add custom code for access from a transmitter switch.

```
 if (RC_Channels::rc_channel(CH_7)->get_radio_in() > 1600)
{ 
         // do something... 
}
```

this way does not work any more in new firmware, how should I get this around?
I don't want to mess with RC_Channel library 
by adding functions like do_function_aux
since it is hard to access core libraries
Thanks in advance.

**Version**
Copter - 3.6.0-rc8

**Platform**
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Any of Copter

**Hardware type**
NAVIO2, SITL

rmackay9, 2018-09-06:
This is really a developer question so best to ask in the dev forums or on gitter... but in any case, I think that should still work.  Another alternative is to use the rc "singleton" like this, "rc().channel(CH_6)-get_radio_in()"

closed 2018-09-06 by rmackay9

SwiftGust, 2018-09-07:
I'm sorry to post thing here I posted this gitter on Research channel but got no response.
And I did try that way too. 
 if (rc().channel(CH_7)->get_radio_in() > 1600)
{ 
         // do something... 
}
But didn't work so I'm looking for another answer to that...

auturgy, 2018-09-07:
@peterbarker is the guy for this one, but CanberraUAV have his focus for the next couple of weeks...

Regards,

James

SwiftGust, 2018-09-11:
This is solved by following way.
In the new firmware, custom code directly inserted in the ArduCopter.cpp / Copter::rc_loop() works fine.
This was used to work with adding custom code in the radio.cpp / Copter::read_radio() function.