ArduPilot/ardupilot #9383: Plane: Fails Boot, Solid yellow GPS LED, No servo outputs

Naterater, 2018-09-07:
### Bug report

**Issue details**

Occasionally boot completely fails. Maybe 5% of the power ups. I get the startup tones from the buzzer and then the GPS lights stay solid yellow and pressing the safety switch has no effect -- I can't get servo outputs. The light around the safety button acts as if it recognizes the safety switch state change because it turns from flashing to solid. It also goes back to flashing when I press the safety button again. A reboot fixes this 99% of the time.

**Version**
Plane 3.9.1 stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Basic electric twin

**Hardware type**
Cube black, Mauch power supplies, RFD900X powered separately

**Logs**
https://www.dropbox.com/s/74on1qw3s2jsn8u/00000145.BIN?dl=0

IamPete1, 2021-07-26:
Is this still a issue on 4.1?

Naterater, 2021-07-26:
I haven't seen it lately.

closed 2021-07-26 by Naterater