ArduPilot/ardupilot #9388: differential spoilers fix- need help compiling with new servo.cpp file to test.

jgilbertfpv, 2018-09-09:
### Feature request - add crow brakes to differential spoilers and fixed so inner moves down/outer up

Differential spoilers have no crow brake option. i've added a fix but get too many errors when i tried to follow instructions on setting up a build environment with cygwin. and i'm on a capped data speed which makes thing very difficult. 

**Describe the solution you'd like**
I've edited servo.cpp and need a px4-v2 and px4-v3 (i think my board is v2 but ardupilot recognizes v3?)

**Platform**
Plane

**Additional context**
[servos.zip](https://github.com/ArduPilot/ardupilot/files/2364668/servos.zip)

Naterater, 2019-01-07:
Looks like this issue was solved in https://github.com/ArduPilot/ardupilot/pull/9391, is that correct?

jgilbertfpv, 2019-01-07:
Yes sorry I forgot about this. I'll close it now

closed 2019-01-07 by jgilbertfpv