ArduPilot/ardupilot #9389: Transmitter off causes brief extreme behavior

Naterater, 2018-09-09:
### Bug report

**Issue details**

I often fly with my transmitter off to conserve batteries on longer missions. When I need to take control, I turn it back on. 95% of flights there is no issue. This flight and one other before had a very nasty behavior -- (60-degree uncommanded roll and pitch down) when I turned off my transmitter. I thought the first time that I noticed this behavior, it was some anomaly. Now I'm sure that it is linked.

**Version**
Plane 3.9.1 stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin-engine believer

**Hardware type**
Cube Autopilot, Mauch BEC's
RFD900X telemetry (I have t.logs as well if necessary)
FrSky Taranis X9D+ and X8R

**Logs**
https://www.dropbox.com/s/w4m0us7ienwcno8/00000171.BIN?dl=0

WickedShell, 2018-09-09:
Closing, this is a support issue. You're receiver is outputting out of range/minimum values as it's failsafe. If you enabled throttle failsafe this wouldn't be a problem.

Naterater, 2018-09-10:
So you're saying this is absolutely a RX/TX problem for not outputting no pulses? That sounds quite dangerous. From what I see, it was outputting way beyond the calibrated range on every channle for about 1 second, and I don't think ardupilot should recognize that as a valid input.

The RX/TX combo here is a common setup, and I still have GCS_FS enabled. Just because I disabled throttle failsafe should not mean there is no issue here. I think there needs to be another THR_FS option where the plane ignores the rest of the channels, but does not trigger FS_ actions. Either that, or give me different FS actions based on which failsafe it is. Battery has its own actions, why don't RC and GCS have their own separate actions?

WickedShell, 2018-09-10:
The throttle failsafe is used to indicate that the receiver is outputting bad pulses. If you had enabled it the pulses would have been ignored. With the failsafe disabled then we must honor whatever your transmitter sends to us.

Naterater, 2018-09-10:
Then there needs to be another option to allow GCS failsafe to trigger returning to home while allowing THR failsafe to continue automatic operation in auto mode (without honoring bad transmitter inputs).

Naterater, 2018-12-31:
Yes, and that works, however it is difficult for me to tell when the TX was not correct in the logs, and if I ever bump a stick as I disconnect, it's difficult to get that channel back to neutral.

IamPete1, 2021-07-26:
We now have continue in auto if in throttle failsafe options. So you can leave the failsafe enabled and choose to ignore it in auto.

closed 2021-07-26 by IamPete1