ArduPilot/ardupilot #9396: SITL: AP_Camera object

khancyr, 2018-09-10:
### Feature request

**Is your feature request related to a problem? Please describe.**
We should add a camera object like for SPRAYER and GRIPPER but for CAMERA.

peterbarker, 2018-09-11:
On Mon, 10 Sep 2018, Pierre Kancir wrote:

> Is your feature request related to a problem? Please describe.
> We should add a camera object like for SPRAYER and GRIPPER but for CAMERA.

tridge's recent work for the OBC will be very useful here - we can produce 
actual images with the camera...

ongspxm, 2019-01-28:
hi, i am looking to pick this up, can i have a rough outline of the requirements for this issue?

> tridge's recent work for the OBC 
regarding this, where will i be able to reference this from?

khancyr, 2019-01-28:
hello,

I think the minimal would be to have a new camera object that is capable to receive malink_camera command, get an input trigger, and output the image from the map.
I think @peterbarker is refering to https://github.com/tridge/ardupilot/tree/CUAV-OBC-2018 .I invite you to look at Tridge post on OutBack Challenge (OBC) on discuss, you will find interesting informations!

marck3131, 2022-01-13:
Hello @khancyr , @peterbarker , As there are no recent messages on this thread , Can you assign this issue ?
I'd like to have a look and work on this. Also please let me know , if this issue is already resolved.
Thankyou.

peterbarker, 2022-01-13:
@marck3131 we can't assign issues from a technical perspective - but feel free to take this on :-)

marck3131, 2022-01-14:
Hello @peterbarker , can you please give me a summary on what actually this issue about? Where do we add CAMERA_OBJECT?
I cannot conclude fairly from the conversations above.
Thankyou.

Sanskriti0805, 2024-10-27:
hello is this issue still open ? can i work on it ?