ArduPilot/ardupilot #9398: Ardupilot fails to boot/upload due to assertion failure in px4io_uploader.cpp:423

arslankhan8, 2018-09-10:
### Bug report
Hi, 

While running arduopilot on pixhawk i added some code to profile how much time we spent in idle mode, i ran into this assertion. On some debugging it looks like the size of firmware uploaded from /etc/px4io/px4io.bin has some alignment requirements. To further track this i tracked where this binary file is coming from. Looking at the code there are multiple places this could come from but since i am not using SD Card it looks like the firmware is coming from the ROM filesystem. By reading Makefile i was able to see that this file is created on the host at $ROOT/modules/PX4Firmware/Build/px4io-v2_default.build/firmware.bin. Further debugging how this is made i was able to find the linker script of the binary, however in linker script ($ROOT/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/nuttx-export/build/ld.script) we don't make any effort to make sure the binary size is 4-byte aligned. 
Seems like px4 firmware uploader expects an aligned binary, however nuttx linker script is not providing any guarantees, which causes the assertion to fail. As a fix i updated the linker script file to make sure the flashed binary's size is 4-byte aligned. For this i added padding by readjusting the location counter in the last section written to flash, which solved my problem. Since this is just a temporary file (build time artifact) i pushed the change to the original file i.e. ($ROOT/modules/PX4Firmware/nuttx-configs/px4io-v2/scripts/ld.script). 

Please let me know if anything else is required from my side. Furthermore i am not sure if this is the right place for this issue but since this is the parent project so i opened it here. Let me know if we need to move this somewhere else.

Thanks,
Arslan

**Version**
I am working with arudocopter v3.3.3. However the px4io_uploader.cpp in head revision has the same assertion, furthermore in the linker script in head revision 

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? Pixhawk
**Logs**
_Please provide a link to any relevant logs that show the issue_
[assert_fail.log](https://github.com/ArduPilot/ardupilot/files/2367836/assert_fail.log)
[fix.log](https://github.com/ArduPilot/ardupilot/files/2367837/fix.log)

**fix/patch**
Here's the patch for the changeset:
diff --git a/nuttx-configs/px4io-v2/scripts/ld.script b/nuttx-configs/px4io-v2/scripts/ld.script
index 69c2f9cb2..385c4f653 100755
--- a/nuttx-configs/px4io-v2/scripts/ld.script
+++ b/nuttx-configs/px4io-v2/scripts/ld.script
@@ -103,6 +103,7 @@ SECTIONS
 		*(.gnu.linkonce.d.*)
 		CONSTRUCTORS
 		_edata = ABSOLUTE(.);
+                . = ALIGN(3); 
 	} > sram AT > flash
 
 	.bss : {

tridge, 2018-09-11:
this looks like a good change, thanks, we just need to PR it to PX4Firmware and PR the submodule update, and test

OXINARF, 2018-09-11:
@arslankhan8 Yes, if you want to, it will be much appreciated. You'll need to fork http://github.com/ArduPilot/PX4Firmware, make the change and open the PR there. Then fork this repo, update the submodule commit and open a PR here.

If you need any help, stop by our dev chat (http://gitter.im/ArduPilot/ardupilot).

WickedShell, 2018-09-21:
The fix is indeed in, and has been for quite awhile.

closed 2018-09-21 by WickedShell