ArduPilot/ardupilot #9400: Poor handling of SBUS Failsafe

R-Lefebvre, 2018-09-11:
FrSky SBUS Failsafe handling is buggy, and Ardupilot needs to protect for the issue.

We have seen undesired mid-flight motor shut downs because of the issue.  Basically, FrSky "no pulses" failsafe actually makes the Rx transmit SBUS "Zero" which our SBUS library converts into the number 874.

There are no protections for this value, even though it is usually below the RC Min in most cases.

I have used this fix for over a year now on many helicopters and it works well:

https://github.com/R-Lefebvre/ardupilot/commit/edb6ef1c9aa3bf28526c186afc950d900a47a272

However, it must also be considered that the FrSky receivers do not always exit a failsafe event "cleanly" because it can take several frames for all the channel to return values in range, as shown here:

https://drive.google.com/file/d/0BzWH4Q7IAU5YeVN0SG1xVEdlVWs/view

If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

tridge, 2018-09-11:
874 is SBUS 0:
>>> SBUS_TARGET_MAX=1000
>>> SBUS_TARGET_MIN=1000
>>> SBUS_TARGET_MAX=2000.0
>>> SBUS_TARGET_MIN=1000.0
>>> SBUS_RANGE_MIN=200.0
>>> SBUS_RANGE_MAX=1800.0
>>> SBUS_SCALE_FACTOR=((SBUS_TARGET_MAX - SBUS_TARGET_MIN) / (SBUS_RANGE_MAX - SBUS_RANGE_MIN))
>>> SBUS_SCALE_OFFSET=int(SBUS_TARGET_MIN - (SBUS_SCALE_FACTOR * SBUS_RANGE_MIN + 0.5))
>>> int((200 * SBUS_SCALE_FACTOR +.5) + SBUS_SCALE_OFFSET)
999
>>> int((0 * SBUS_SCALE_FACTOR +.5) + SBUS_SCALE_OFFSET)
874
>>> int((1000 * SBUS_SCALE_FACTOR +.5) + SBUS_SCALE_OFFSET)
1499
>>> int((1800 * SBUS_SCALE_FACTOR +.5) + SBUS_SCALE_OFFSET)
1999
>>>

tridge, 2018-09-11:
unfortunately it is possible to send SBUS 0 (ie. 874) as a valid output on one channel while other channels are normal

olliw42, 2018-09-11:
sbus has a dedicated failsafe flag, maybe its useful ;)
naively I also would guess that if there is one failsafe 0 that all channels are 0 (if the failsafe flag doesn't cover that already)

SergeyBokhantsev, 2018-09-11:
I had the same issue with X8R https://github.com/ArduPilot/ardupilot/issues/7516

Nurffe, 2018-09-27:
Probably somewhat same issue but other way around can be seen here where failsafe is triggered. Failsafe is set to "no pulses" but still zero is sent for certain channels (5, 9, 10, 11, 12, 14). There is no preset failsafe set in receiver itself so it's unknown why these random channels are set to zero which is then converted to 874 in FC

Here are snapshots from logs

Showing lines and timestamps for radio fs trigger
[image]

Showing lines and timestamps for incorrect channel values
[image]

Error state only lasts ~400ms and happens shortly after failsafe is triggered. Seeing this happen *after* failsafe has been recognized by FC makes me believe there might actually be something in arducopter code itself and not just receiver doing weird things. Ardu shouldn't take nonsense inputs when failsafe flag is already on

rmackay9, 2019-01-11:
This is fixed in master and Copter-3.6.5 which is going out for beta testing today.  Thanks for the report!

closed 2019-01-11 by rmackay9

rmackay9, 2019-01-14:
@Nurffe, [this is the fix that went into Copter-3.6.](https://github.com/ArduPilot/ardupilot/pull/10216).  It's a similar but slightly different fix for master because the auxiliary switch code has been moved into the RC_Channel object in master.