ArduPilot/ardupilot #9404: mini-pix : internal magnetometer orientation

mric3412, 2018-09-11:
### Bug report

**Issue details**

After compass calibration, mx corresponds to (-my) and my corresponds to mx. You have to define the internal compass as external in order to affect compass orientation at YAW270 (instead of none)
With Yaw270, m(x,y,z) are correct for internal compass. Is it an hardware problem or a wrong management of this specific board in arduplane v3.9.1 ?

**Version**
arduplane v3.9.1 stable

**Platform**
Plane

**Airframe type**
flying wing

**Hardware type**
mini pix from radiolink

mric3412, 2018-09-12:
I updated the firmware via MissionPlannner supposing GCS will detect the specific board and propose the proper firmware. Actually, it was not the specific firmware for mini-pix.
No issue with magnetometer with 3.9.1 builded for mini-pix (download the proper .apg online and load custom firmware)
A popup message from GCS before uploading the firmware would be good to avoid such confusion.

closed 2018-09-12 by mric3412

yo8aiv, 2020-07-11:
wich is proper apj. .. i'm having same issue here. thanks.