ArduPilot/ardupilot #9407: Rover: add support for electric trolling motors

rmackay9, 2018-09-12:
It would be nice if we added support for controlling the speed and direction of electric trolling motors like [these from Minn Kota](https://minnkotamotors.johnsonoutdoors.com/freshwater-trolling-motors).  It is likely that these motors accept a PWM at least for the motor speed and possibly for the servo input as well.

There are many other low cost trolling motors on eBay as well that may also work.

This task involves:

- finding a commonly available motor which accepts a pwm input for motor speed and ideally motor direction as well (i.e. motor includes a servo to move the motor).
- testing that it works with ArduPilot and/or modifying ArduPilot to support the motor
- document the connection and setup on our Rover/Boat wiki

This request first came up [here on discuss](https://discuss.ardupilot.org/t/newbie-questions-about-boat-frames).

IamPete1, 2021-08-29:
We support Torqeedo now, https://github.com/ArduPilot/ardupilot/pull/17979 so maybe we can close?

rmackay9, 2021-08-30:
yes, we can close this now, thanks!

closed 2021-08-30 by rmackay9