ArduPilot/ardupilot #9408: Copter: add support for precision landing using gimballed camera

rmackay9, 2018-09-12:
Copter's precision landing assumes that we the camera is hard mounted to the bottom of the vehicle.  At least one users has requested that we support the hardware setup where the camera is mounted on a roll/pitch gimbal.  This means the yaw will still rotate around with the vehicle but the camera will remain flat.

fnoop, 2018-09-12:
This has been asked numerous times in the past - would be great to add with 3-axis freedom.  It would allow anyone to use a gimballed camera for other purposes in normal flight and then use it for precland when coming into land.  Would be really cool if we could control the gimbal to follow the target as it gets closer, or do scans/sweeps for landing targets.
It would require a reasonable effort to make it possible.  I suck at the necessary maths but if no-one else has picked it up when I clear my todo list I'll have a look.

auturgy, 2018-09-12:
@tridge possibly relevant to gimbal discussion.

Regards,

James

rmackay9, 2018-09-19:
@fnoop, what do you think about this change to support a camera on a gimbal?
https://github.com/rmackay9/rmackay9-ardupilot/commit/c824c2ea7ae76dc5235420b9bb05585d382ad6dc

jschall, 2018-09-19:
> rmackay9@c824c2e

Pretty simple, should be one parameter.

philippewarren, 2023-09-02:
We would like to use this feature for our project. What is the status of the development on this? I see that there was a relatively simple commit adding support a few years ago (https://github.com/rmackay9/rmackay9-ardupilot/commit/c824c2ea7ae76dc5235420b9bb05585d382ad6dc). Was it complete and tested? Was it merged? What work would be required to complete this feature today?