ArduPilot/ardupilot #9415: IRLock Support for QuadPlane

jonathan84clark, 2018-09-12:
Hello,

I was wondering what it would take to get IRLock support for quadplanes? I have started the process it seems like its a little more involved then simply including the driver. Is this true? Is support for this device planned anytime soon?

Thanks,

Jonathan L Clark

rmackay9, 2018-09-13:
Jonathan,
Yes, there's more to it than that unfortunately.  Beyond the sensor driver (AC_PrecLand, AP_IRLock) there will need to be some changes to the quadplane's landing code to use IR_Lock.  Copter will have some code that can be copy-pasted across but I'm sure you'll find that the structure of plane and copter are quite different so it's a bit of a challenge to figure out where to put it in Copter.  By the way, we are hoping to restructure Plane over the next 6 months or so to make it more consistent with how Copter, Rover and Sub are structured.

I might be good to try and get some help from the Plane experts like Tridge, Michael DB (aka wickedshell) or Tom Pettinger (aka magicrub).  Either show up in Mumble and/or ping on the gitter chat.

jonathan84clark, 2018-09-13:
@rmackay9
Thanks for the info. I'm pretty new at this stuff. Just out of curiosity. Your the expert on this. How long do you think it would take you to implement this change?

rmackay9, 2018-09-13:
@jonathan84clark, I think for a plane expert it's probably a couple of days effort for the initial development (i.e. not including testing on a real vehicle) but that's a guess.  For me personally, I wouldn't attempt it without first doing the restructure but that's just me... that requirement wouldn't apply to others probably.

jonathan84clark, 2018-09-14:
@rmackay9 
Thanks again for the info. 
I have started adding the feature to arduplane. I have a project that requires IR-LOCK for quad-plane within the next two weeks. So far things seem to be going well but I am running into a compile error when I call precland.init()
`/home/jonathanc/arduJonathan2/ardupilot/build/px4-v3/../../libraries/AC_PrecLand/AC_PrecLand_IRLock.cpp:11: undefined reference to `vtable for AP_IRLock_I2C'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.`

I am guessing that I am missing a library or resource however I am still pretty new to the ArduPilot build system and I was wondering if you have any recommendations on this?

(Also I have created a fork for ArduPilot on my personal repo https://github.com/jonathan84clark/ardupilot I am working in the irLockQuadPlane branch). All of my work has been confined to the files in ArduPlane/

Thanks,

Jonathan L Clark

jonathan84clark, 2018-09-14:
Never mind I figured it out!
I had to include IR_Lock in both the wscript and include it and precland in make.inc
Now it is compiling again.

jonathan84clark, 2018-09-28:
Started testing my changes. The problem is that for some reason the IR-LOCK sensor never communicates with the Pixhawk 2.1. I tried all three I2C buses (default, internal, external). 
I connected the IR-LOCK sensor to the I2C port in the center of Pixhawk 2.1 (not the GPS port). 
I read on one form that I may need to "hack" the IR-LOCK driver because it has a hard-coded I2C port. Don't know if that is an issue.
Any help on this would be great!

Thanks,

Jonathan L Clark

jonathan84clark, 2018-10-02:
Okay I figured out the issues.
1. When connecting the the Pixhawk 2.1 I2C port (the one in the middle of the board). The PLND_BUS needs to be set to 0. (I never saw this spelled out anywhere so here it is!)
2. The Pixy needs to have a special version of the firmware on it to work as an IR-LOCK sensor. https://irlock.readme.io/docs/firmwaresoftware-links

I flight tested my code and we were able to do four precision landings in high winds and on a slope. More testing will be required to test in autonomous flight. Hopefully I am close to doing a pull request.

ddomit, 2019-12-04:
> Okay I figured out the issues.
>
> 1. When connecting the the Pixhawk 2.1 I2C port (the one in the middle of the board). The PLND_BUS needs to be set to 0. (I never saw this spelled out anywhere so [...]

Hi! Is this already in the Plane code? If not how can we add it? @rmackay9

vtolworx, 2020-05-10:
Can you post your compiled firmware? I want to start testing this!!

Ryan

ddomit, 2020-08-19:
Hello Guys!

Any news on the integration of this feature to VTOL?

ddomit, 2020-08-19:
> Okay I figured out the issues.
>
> 1. When connecting the the Pixhawk 2.1 I2C port (the one in the middle of the board). The PLND_BUS needs to be set to 0. (I never saw this spelled out anywhere so [...]

How did your tests go?

ohitstarik, 2021-04-01:
Kind of wanted to bring attention to this again.

This has been (successfully) attempted by others but there's no stable implementation into ardupilot master, it's always fallen through the cracks at the last stages.

Theres two previous attempts that I wanted to bring back to light and work on, but for some reason none of them work for me. #10378 compiles and runs, but never gets a target lock (or, logs never indicate a target lock. This is with irlock cam). Arducopter from same firmware gets target lock no problems. #11339 builds, but the cubeBlack doesnt boot properly/run properly/send any mavlink messages. Thank you to these two devs for their work on it so far.

edit: Theres a 3rd implementation as well: #13263. Just tried it and logs still dont show a target lock, compared it to the arducopter from the same firmware and that one shows target lock.

jonathan84clark, 2021-04-01:
I really wanted to finish this feature (get it pulled into the main code) back in 2018. But I ran into some major scope creep. The community kept piling on things they wanted me to roll into this pull request; refactors, minor code changes, bug fixes, etc. The precision landing feature for quadplane was functionally completed. I validated that it worked on a real UAV and the company I was working for is still using precision landing (as I implemented it) last I checked. Unfortunately that company laid me off in late 2018 so I didn't have access to any UAVs to test additional code changes, that pretty much stopped me dead in the water. And thus, I couldn't implement all of the (scope creep) features that were asked of me. 

I suspect the company I worked for still has a copy of my code although at this point I doubt they have the resources to attempt another pull request. Especially since that would require rolling  in a whole bunch of other changes.

Myro90, 2021-09-28:
Hi, does anyone have the lastest information on whether the IR-Lock sensor and MarkOne beacon system has improved on quadplane (VTOL)

peterbarker, 2024-07-31:
We have this support now

closed 2024-07-31 by peterbarker