ArduPilot/ardupilot #9416: Enhancement: NTF_BUZZ_ENABLE extra setting

xfacta, 2018-09-13:
### Feature request / Enhancement
I’m requesting an extra setting for the parameter **NTF_BUZZ_ENABLE**
Currently there is just Enable or Disable (1 or 0)

**Could we add a “disarmed only” or “not inflight” (value 2) setting ?**

- This would have the effect of only allowing buzzer activity while the FC is disarmed or landed.
- The buzzer would remain inactive/silent while the FC is armed/flying

**Reasons:**

- Small FC’s like the Pixfalcon have the buzzer built in, during flight the buzzer can produce altitude jumps caused by vibrations (and maybe other effects) when it beeps for things like flight mode changes.
- The buzzer is impossible to hear during flight, so it may as well be inactive
- The buzzer is useful for audible status updates while the FC is disarmed/landed

**Platform**
[  ] All
[  ] AntennaTracker
[Y] Copter
[Y] Plane
[  ] Rover
[  ] Submarine

This could possibly apply to all platforms, but I presume Copter and Plane would be the most affected by vibrations from the buzzer

ariefadha, 2021-09-16:
is this still open?

I have feature request for buzzer too, 

**Could we add more buzzer notification**

- This might help to tell pilot when pre-arm ready and gps 3d locked

**Reason**

- I'm user with goggle. when in the field, waiting for pre-arm ok and 3d gps ok, i have to look back to the goggle, and it drain my gogle batre. so i put oled to notify me, but i have to remove oled and put the hood before flight. If buzzer can notify when arming ready is very useful

xfacta, 2021-09-20:
The best thing is to mount an external RGB LED module so it's easily visible
This gives good visual indication and is often visible while flying too
Something like this on ebay, but there's thousands of others out there

https://www.ebay.com/itm/Pixhawk-PX4-RGB-External-LED-USB-Extension-Module-Full-Color-w-Protective-Case-/283435027584