ArduPilot/ardupilot #9435: AP_Notify rewrite misconfigures on Navio2

votingprawn, 2018-09-17:
Hi,

I believe that a recent rewrite of AP_Notify ( 961e16dca96ef2188f198b5a22328c5610393f29 ) has introduced a bug that results in misconfiguring the LED when compiled for Navio2 boards.  I think there is an elif missing around lines 224-234 that is resulting in 'additional' backends being set for the Navio2.

https://github.com/ArduPilot/ardupilot/blob/0e0d847f7f8b1a3f79ca173ccccb919345b69190/libraries/AP_Notify/AP_Notify.cpp#L208-L234

When you compile after this commit (or download any build from after this commit) it appears to run fine but fills the console with

``GPIO_Sysfs: Unable to write pin 16 value.``
``GPIO_Sysfs: Unable to get value file descriptor for pin 4.``

I can fix the issue for myself by commenting out 224-234, but I'm lacking the mental acuity this morning to unpick the if statements to work out exactly whats going wrong.

auturgy, 2018-09-17:
> I believe that a recent rewrite of AP_Notify ( 961e16d ) has introduced a bug that results in misconfiguring the LED when compiled for Navio2 boards. I think there is an elif missing around lines 224-234 that is resulting in 'additional' backends being set for the Navio2.

@wickedshell

WickedShell, 2018-09-17:
@votingprawn It looks like an AP_BoardLED is being added since the defines for `HAL_GPIO_A_LED_PIN` `HAL_GPIO_B_LED_PIN` HAL_GPIO_C_LED_PIN` are all set. If there aren't any board GPIO led's on a Navio2 we should not be setting those defines in https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/board/linux.h#L149

Right away though we can see an obvious bug here in that all 3 pins are defined to be the same value. We can easily make `undef`'s but I don't have a schematic for a NavIO2. Are there any GPIO controlled LED's we should be using, or should these define never have been set?

votingprawn, 2018-09-17:
@WickedShell thanks for your help.  There is one LED on the NAVIO2 which _is_ pinned out on GP4, GP27, and GP6, but my understanding is its already being addressed above by LED_sysfs?

https://github.com/ArduPilot/ardupilot/blob/0e0d847f7f8b1a3f79ca173ccccb919345b69190/libraries/AP_Notify/AP_Notify.cpp#L210

WickedShell, 2018-09-17:
@votingprawn That matches my guess as well. I've put in a PR #9440 which should correct this, as well as catch obvious bad pin assignments. Would you be able to verify that it fixes the problem for you?

PR #9440 (2018-09-19): Remove BoardLED from NavIO2
This resolves #9435, NavIO2 does not have any board led's, they are all controlled via Led_Sysfs. This also adds static asserts to the board leds to catch the obvious overlap of pins, which can't ever be a valid configuration.

@votingprawn would you be able to test that this resolves the issue you were seeing?
closed 2018-09-19 by OXINARF