ArduPilot/ardupilot #9442: Fast takeoff in poshold mode to midair then continue small shift can't hold its position

askariz, 2018-09-18:
### Bug report

**Issue details**

Fast takeoff (First let PILOT_SPEED_UP to 10m/s then full throttle takeoff in poshold mode) in poshold mode to midair then continue small shift can't hold its position !
[image]

And I find it due to code in **mode_poshold.cpp** there are 3 call for **loiter_nav->init_target();**
[image]

**Version**
copter-3.6/master

**Platform**
[  ] All
[  ] AntennaTracker
[ ✔ ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
hex copter

**Hardware type**
SITL & PX4-V2

**Logs**
https://drive.google.com/file/d/1dChBmx2aOMKALWHLzCjgYflMhGAB6Wgg/view

lthall, 2018-09-18:
Sorry but it is not clear what you are saying the problem is. I can't even see what you have plotted in your graph.

Could you please attempt to describe this problem in more detail.

lthall, 2018-09-18:
Ok, I think I see what you are saying and I think I have fixed the issue.

Could you test this?

https://github.com/lthall/Leonard_ardupilot/tree/20180918-PosHoldFix

Thank you very much for bringing this to our attention!!

askariz, 2018-09-18:
HI , I have tested your commit with SITL then the problem solved . 
Thank you .

rmackay9, 2018-09-18:
fix merged into master and will go out with -rc11.

closed 2018-09-18 by rmackay9