ArduPilot/ardupilot #9452: Parachute Enable Option Causes Parameters to fail to be read fully.

eovnu87435ds, 2018-09-19:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
_Please describe the problem_
When configuring a parachute, if both "parachute enable" and "parachute release" are set on RC channel options, on the next reboot. GCS will not be able to fully read all parameters. Mission Planner, APM Planner 2, and QGroundControl all fail to read all parameters, hanging on VISO_ORIENT. This happens regardless of connecting via USB or telemetry modem. If I remove "parachute enable" from the RCx channel option, on reboot all parameters read fine again. 

Here's the relevant parameters I used
CHUTE_ALT_MIN        3
CHUTE_DELAY_MS     300
CHUTE_ENABLED       1
CHUTE_SERVO_OFF   1050
CHUTE_SERVO_ON    1950
CHUTE_TYPE              10

SERVO10_FUNCTION  27
SERVO10_MAX            2000
SERVO10_MIN             1000
SERVO10_REVERSED    0
SERVO10_TRIM            1500

CH10_OPT    21     (Chute enable: if this is set params don't load)
CH11_OPT    22     (Chute release)

**Version**
_What version was the issue encountered with_
Copter 3.5.7 Stable
Mission Planner 1.3.58 Winx64
APM Planner 2.0.24 Winx64
QGroundControl 3.2.4 Winx64
**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
OctaQuad
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 2.1
FrSky R-X8R connected via S.Bus
**Logs**
_Please provide a link to any relevant logs that show the issue_

JacksonUAS, 2019-10-06:
Is this issue still open???!

rmackay9, 2019-10-07:
@JacksonUAS, it's still open and it doesn't look like anyone has looked into it yet so I expect the issue is still there in Copter-4.0.0 as well.  Hopefully we will get to it.

IamPete1, 2021-07-26:
The `CH10_OPT ` param causing the issue is gone in 4.1, I have been playing with parachutes recently and never hit this, so I'm going to close.

closed 2021-07-26 by IamPete1