ArduPilot/ardupilot #9456: 4 servo swash plate support

maxdrougge, 2018-09-20:
#### Issue details
_Please describe the problem, or desired feature_
I have a helicopter which utilizes a 4 servo swashplate. Currently I'm using some experimental code which supports a 4 servo swash plate. The code works, but the problem is that it is not compiled or implemented in a way that it works iwth Pixhawk v2, it only works with a very old version of pixhawk. 
So, my request is that the 4 servo swashplate code is added to the master-branch so that it will be supported by pixhawk v2 hardware. The code I currently use can be found here: https://github.com/ArduPilot/ardupilot/pull/3245

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Copter, 4-servo swashplate
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk v2
#### Logs
_Please provide a link to any relevant logs that show the issue_

bnsgeyer, 2018-09-20:
@maxdrougge this is being discussed.  @ChristopherOlson and I intend to get addition swashplate support into 3.7.  I'm sorry that it doesn't meet your need immediately.

ChristopherOlson, 2018-09-20:
I have WIP code for this for 3.7, as well as major updates to how swashplates are defined. We'll likely have some test builds back-ported to 3.6 so they can be flown and tested prior to merging in master.

maxdrougge, 2018-09-21:
Thanks for the quick response! Do you have an ETA when the code will be released? If you would like to share the WIP code which has support for 4 servo swash plate I'd be willing to test it out.

ChristopherOlson, 2018-09-21:
It will be sometime before the end of the year (hopefully). There is other changes going into 3.7 that have to be done first. I do not have flyable code at this time. I have the four-servo mixer written, have not implemented the servo support for it yet. Bill and I have to discuss that more to decide which servo goes to the fourth swashplate servo, and do we move the rudder servo to SERVO5, leave it on SERVO4 and use SERVO5 for fourth swashplate servo, etc.. Combined with this is going away from the servo position params and switching to use of pre-defined swashplate types (the four servo is H4-00, as an example). The current code his hard-wired for only three-servo control. I will post here when I get flyable code that can be tested.

maxdrougge, 2018-09-24:
Alright. Please share it when you have some test-code ready. Will you add support so that you can change the angle at which the servos are placed. That is if my first servo is at 45 degrees turned clockwise compared to the H4-00?

maxdrougge, 2018-09-28:
Hello again. 
Just wondering if this code will support this flight controller: 
https://docs.px4.io/en/flight_controller/pixhawk-2.html 
If so, will it also support the FMUv5, pixhawk 4?

bnsgeyer, 2018-09-28:
It ahould support all of the controllers that Ardupilot supports.

bnsgeyer, 2019-06-22:
4 servo swashplate support was incorporated into master (3.7-dev) in PR #10469.  This can also be found in ArduHeli for Copter 3.6 at this link [Four servo binaries](https://github.com/ChristopherOlson/ardupilot/releases/tag/v3.6-Dev)

closed 2019-06-22 by bnsgeyer