ArduPilot/ardupilot #9458: Can't compile with waf - "CMakeCache.txt is created more than once"

maciek252, 2018-09-20:
A fresh install on ubuntu, fresh clone from github. For every vehicle type I've tried, and also for the examples, the following problem appears:

 Node /home/maciek/REPOS/svn/mackowe/drony/ArduPilot/Ardu/ardupilot/build/px4-v2/modules/PX4Firmware/CMakeCache.txt is created more than once (full message on 'waf -v -v'). The task generators are:
  1. 'px4_msg_gen' in /home/maciek/REPOS/svn/mackowe/drony/ArduPilot/Ardu/ardupilot
  2. 'examples/BARO_generic' in /home/maciek/REPOS/svn/mackowe/drony/ArduPilot/Ardu/ardupilot/libraries/AP_Baro/examples/BARO_generic
If you think that this is an error, set no_errcheck_out on the task instance

-----

I can build with make the copter version. But I'm interested in building the examples.

matteoscanavino, 2018-11-02:
Hello maciek252, 

I have the same problem with Ubuntu 16.04... Do you have any suggestion?

maciek252, 2018-11-04:
Hi, I compile with target fmuv3, waf works fine there.

matteoscanavino, 2018-11-05:
Thank-you maciek252. I still have this problem (even with fmuv3):

* Node /home/matteo/ardupilot/build/px4-v3/modules/PX4Firmware/CMakeCache.txt is created more than once (full message on 'waf -v -v'). The task generators are:
  1. 'px4_msg_gen' in /home/matteo/ardupilot
  2. 'bin/arducopter' in /home/matteo/ardupilot/ArduCopter
If you think that this is an error, set no_errcheck_out on the task instance

I'm not sure the compilation is succesful or not as at the end it got: 

Build commands will be stored in build/px4-v3/compile_commands.json
'copter' finished successfully (17.742s)

auturgy, 2018-12-16:
I’ve just run through a bunch of tests and it is working for me (Ubuntu 18.04). I have 2 things I’d like you to do:
1. make sure the sub modules are updated (git submodule update —init —recursive —force)
Then try building the examples.
2. If that doesn’t work, type “gcc -v” to see what version of the compiler is installed, and paste the bottom line of the output in here.

OXINARF, 2019-06-11:
We don't support PX4 HAL anymore so I'm closing this.

closed 2019-06-11 by OXINARF