ArduPilot/ardupilot #9461: Waf: unable to build on windows using cyqwin

IamPete1, 2018-09-20:
### Bug report

**Issue details**
Unable to build master using waf in cyqwin. get linking error: 

Linking build/CubeBlack/bin/arducopter c:/progs/gnu_arm/2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file /cygdrive/d/projekte/Multicopter/ardupilot/build/CubeBlack/ldscript.ld: No such file or directory collect2.exe: error: ld returned 1 exit status

as found by @amilcarlucas and myself, it is independent of build vehicle and flight controller

Manged to track issue to somewhere between 17b61f72a30ae7a81524468ff51559ccfe1b0ecd and 56afc44c56bb334617708cb181e005d994b4a02b . 

this is a attempted fix by @khancyr , but no luck unfortunately https://github.com/ArduPilot/ardupilot/pull/9457

Thanks to all on gitter who helped diagnose 

**Version**
master 

**Platform**
[ - ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

OXINARF, 2018-09-20:
This is caused by this change: https://github.com/ArduPilot/ardupilot/pull/8406/files#diff-443e8eb8ec56b02fb08133fd9e0dc667R401

I don't see any reason for the ld script needing an absolute path now, but @tridge or @bugobliterator will have to say.

IamPete1, 2018-09-20:
I have just tested this and can confirm that changing that line back to the old one fixes the issue

Thanks @OXINARF

commit 434afa26 (2018-09-21): waf: change ChibiOS ld script path back to relative

Fixes #9461
closed 2018-09-21 by OXINARF