ArduPilot/ardupilot #9472: Support for new MavLink v2.3 RC_CHANNELS_OVERRIDE (18Ch Extension)

careyer, 2018-09-24:
Link to discussion in the ArduCopter forum: https://discuss.ardupilot.org/t/support-for-new-mavlink-v2-3-rc-channels-override-18ch-extension/33319

### Feature request

**Is your feature request related to a problem? Please describe.**
Dear developers,

afaik ArduCopter 3.5 onwards supports the overhauled MavLink v2 protocol (as of ArduCopter 3.6 the MavLink v2 protocol is considered default). Recently in MavLink v2.3 an extension for RC_CHANNELS_OVERRIDE supporting up to 18 channels was standardized
( https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE 4 )
Up until now ArduCopter - although supporting MavLink v2 - does only support 8Ch.

We are really struggling to get this to work. We’ve setup a small test environment and compiled the latest MavLink v2 library on a ESP32 (we did so to have better debugging possibilities). From there we are generating according RC_CHANNEL_OVERRIDE MavLink messeages and feed them into a PIxhawk running the latest ArduCopter developer build (serial port protocol is set to Mavlink v2).
We can happily see the Channels 1-8 moving in mission planner. However Channels 9-18 show no reaction.
We also checked with a logic analyser that the messages are correct and indeed channels 9-18 are transfered to the Pixhawk correctly. However they do not get interpreted

**Describe the solution you'd like**
Please support the up to 18Ch RC_CHANNELS_OVERRIDE Extension standardized with the MavLink v2.3 message extensions: https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE 4

**Describe alternatives you've considered**
There is no alternative for professional applications than having more than the currently supported but very limited 8CH via GCS telemetry (RC_CHANNELS_OVERRIDE). Only 8 channels are consumed quite quickly when it comes to professional applications. The upgrade to either 16 (Mission Planner is ready for that already) or 18 channels would yield a huge improvement with considerably minimal effort. Seems to be a low hanging fruit. 

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE 4

THANK YOU VERY MUCH AND KEEP UP THE EXCELLENT WORK!!!!!
Best regards

Thomas Berndt

htcohio, 2018-09-26:
This would be really helpful for our project.  It would be greatly appreciated if this could be addressed ASAP because we're working on an agricultural survey and spectral Imaging platform where several controls need to be accessible via mavlink in our app.

18 channels would be nice but 12 to 16 channels would be sufficient if that would be easier.

dino-de, 2018-10-02:
I agree, at least 16 channels would be a nice to have!

Yes21, 2018-10-05:
I also agree with this feature request. The break of the 8 channels limitation would be a great enhancement to ArduPilot, and would allows more applications.

Thanks for this great job !!

user1321, 2018-10-05:
That kind of feature will move platform to whole new level of functional and significantly increase range of possibilities.
We definitely need it as soon as possible.

peterbarker, 2018-10-05:
On Fri, 5 Oct 2018, user1321 wrote:

> That kind of feature will move platform to whole new level of functional
> and significantly increase range of possibilities.
> We definitely need it as soon as possible.

OK, so now I'm curious.... what's the use case for using 
RC_CHANNELS_OVERRIDE which makes this such a game-changer?

novaent, 2018-10-05:
It would be a great idea to have 16 channels through WBC-RC control. 4 channels are required for camera control. tilt, pan, zoom , recording on/off are essentially required. this way no additional RC set is required for camera control.

marcdornan, 2018-10-06:
> On Fri, 5 Oct 2018, user1321 wrote: That kind of feature will move platform to whole new level of functional and significantly increase range of possibilities. We definitely need it as soon as [...]

EZ WiFi Broadcast supports rc control as well as HD video. 8 channels is very limiting. We really need the full Monty here to make full use of this feature.

careyer, 2018-10-06:
Yes, this is also my usecase.

**Introduction of usecase:**
Let me first introduce to everyone what "EZ-WiFibroadcast" (EZ-WBC) is: EZ-WBC started as a DIY digital wireless HD-Video transmission system (uni-directonal). However over the course of ~3 years it has much matured and today it is an all-in-one bi-directional HD-Video+Telemetry+Control solution (very similar to DJI Lightbridge in a sense). Speaking of Lightbridge EZ-WBC is open source, way more flexible and much greater range can be achieved (currently 31km is the setpoint).
You can learn more about the project here: https://www.rcgroups.com/forums/showthread.php?2664393-EZ-WifiBroadcast-cheap-digital-HD-transmission-made-easy!/page243
The pure length of the thread shows that there is huge interest and demand for this.

**How does EZ-WBC work in conjunction with AurduPilot?**
EZ-WiFiBroadcast offers a high bandwidth low latency bi-directional telemetry link (MavLink) to the airscraft. Via this it can also take over RC-control via the RC_CHANNEL_OVERRRIDE Mavlink messages. We are using this feature for about 2 years now and it works flawlessly. The only bottleneck was the 8Ch limitation of this function in  MavLink v1.... We happily observed that MavLink v2 intruduced the above mentioned 18Ch channel extension and immediately integrated MavLink v2 support to EZ-WBC. Sadly we had to learn that ArduCopter despite of supporting MavLink v2 as a default as of Copter 3.6 does not yet integrate this feature. 

@rmackay9: May I kindly bring your interest to this? I'm going to join the previous speakers: This would be an awesome addition (and I believe not too hard to implement) for the next major Copter 3.6 release for sure. Excuse me: I am not too familiar with GutHub - do I have to tag the topic myself or can you add the suggested tags maybe: Mavlink ; Enhancement; Copter; Library ?

careyer, 2018-10-12:
@auturgy : Thank you very much for taking the effort to "tag" this feature request. Is there a reasonable chance that it might get implemented in one of the next releases? It would be AWESOME :boom:

khancyr, 2018-10-12:
Hello,
I don't really like the rc_override because it is really error prone. 
If you get a mavlink stream, why bother with rc_override when we have much richer message for quite everything?

careyer, 2018-10-12:
@khancyr : Thank you for yor objection. What problems did you encounter with RC_CHANNEL_OVERRIDE? We are using it for more than two years now in conjcution with the afforementioned EZ-WBC. It has always worked flawlessly and allowed for breathtaking flights. Please notice: We do not run it in parallel with a regular RC receiver. We are sole trusting in RC_CHANNEL_OVERRIDE as the one and only control system. It never gave us any problems. What are your experiences?

dposea, 2018-10-23:
I submitted a PR for Copter that  adds the extra channels. It has been tested but only by 1 person so far. It's my first PR so I apologize if I did not get everything correct.

specyfick, 2019-04-28:
Is there any change in this enhancement ?
Need it for rover.

htcohio, 2019-04-28:
Hi, yeah is there some reason that this has not been implemented yet? It
seems logical that it should be especially with all of the advancement in
companion computers with the pixhawk such as EZ wifibroadcast, TengriUAV,
open.HD and other systems...

khancyr, 2019-04-28:
it is already in master, it should come with copter 3.7 and rover 3.6

htcohio, 2019-04-28:
Oh really???

htcohio, 2019-07-21:
Hey guys,

Copter 3.9 does not appear to be working with ch 9-16 as stated.

For mavlink 2 RC with RCOverride only channels 1-8 appear to be working...

Any updates on this?

(See pic)
[image]

magicrub, 2019-07-21:
Closing this. It's implement in master for many months and is probably included in a make releases done this year. The problem you're having is mission planner doesn't have the GUI to configure it yet. That's a mission planner problem. The parameters are all there in Ardupilot.

closed 2019-07-21 by magicrub

htcohio, 2019-07-22:
Hello,

Unfortunately this does not appear to be working.

We cannot utilize channels 9-16 like we can on Dposas modified firmware from last year.   I first loaded that firmware and you can see that channels 9-16 both work with a servo or output.

Then, on the official arducopter 3.6.9 we see no output on Mission planner or at the servo/channel. 

Any ideas?

P.S. Here is a Gdrive link to the modified dposea firmware.
*
The "V2" plane/copter was for regular Pixhawk and "V4" plane/copter was for Pixracer FW. (Or Vice-Versa) I forget which one went to what, but they are only compatible with their respective boards. 

https://drive.google.com/folderview?id=13nKY-nrxFuK3xrRaHFRZRNW0hFG3J28M

[image]
[image]

magicrub, 2019-07-22:
As stated, it's been fixed in the master branch and has been in major releases this year. However, copter has not had a movie release, only minor 3.6.x ones. It will be in copter 3.7.0 or you can take a look at the beta.

careyer, 2019-07-22:
@magicrub : I just tested the issue @htcohio is claiming. I tried with both the latest
- ArduCopter 3.7dev 
- ArduPlave 3.9.9beta 2

On both Channels Ch9-Ch16 are confirmed to not work. (not displaying anything in Mission Planner, nor outputting any signal to a connected servo).
Please don't me wrong: I totally trust you when saying that the code was included but maybe there is a bug preventing it to work properly or we missed that the mimic in using RC_CHANNEL_OVERRIDE has changed somehow? I wonder: Was it successfully tested by someone?

P.S: Cross-Checked my setup with the initially modded firmware by user @dposea. (@htcohio linked them above). Those are working flawlessly.  I think that means my test-rig is working at least.

May I ask you to have another look at the code that was added and test if it actually works? Thank you Sir!

Best regards

rmackay9, 2019-07-22:
maybe the SERIALx_PROTOCOL needs to be set to "2"?  It's just a guess..

careyer, 2019-07-22:
@rmackay9 : Check! We tried this already as it was also our first guess!

peterbarker, 2019-07-22:
On Mon, 22 Jul 2019, careyer wrote:

> @rmackay9 : Check! We tried this already as it was also our first guess!

I've just tested on a Cube running master and it works just fine using a 
patched MAVProxy.

How were you testing?

I did find MAVProxy wasn't doing well in terms of RC channels overrides; 
I've created a fix here:

https://github.com/ArduPilot/MAVProxy/pull/648

careyer, 2019-07-22:
Allright we were able to find the problem. When updating Copter 3.6x to 3.7dev some old settings that do not get overwritten during the flash procedure cause this problem. We had to flash a completely different firmware (e.g. Rover) first and then revert to the Copter firmware. That cleared all settings and now it is working! Thank you very much!

astragalizont, 2020-12-12:
Hi, is this generally know to work with 12-16 channels? I am Using CRSF RC over Mavlink and can only get 8 channels, even when 12 are activated in the Transmitter and reciever. I am using a F405 Wing and RX TX from the reciever to telemetry port 1 and a CRSF diversity reciever, and mavlink 2 on serial 1 (telemetry 1)