ArduPilot/ardupilot #9474: Airborne home tracker for directive antennas (extending range)

mric3412, 2018-09-25:
Antenna tracker already exists for the GroundControlStation.
Having two servos pointing directional(s) antenna(s) from plane towards GCS/HomeWP would further extend the range (or optimize the airborne Tx power )

This could be done by adding 2 functions to the channels ouput functions:
1- PAN_TRACK_HOME
2- TILT_TRACK_HOME

Users will then use a 360deg servo controlled by the pan_track_home output
and a 180deg servo controlled by the til_track_home output
in order to point the aircraft's antennas to Home or GCS (these antennas can then be directional, which will save transmission power and/or extend the range and generate less RF interferences)

PS : in most cases, Home is a fixed waypoint defined when arming. For specific applications, the GCS could be mobile and a telemetry link could be used to use a dynamic home waypoint.

Platforms : Plane, Copter and Rover

IamPete1, 2018-09-25:
I think you can do this in copter with a servo gimbal http://ardupilot.org/copter/docs/common-camera-gimbal.html Just set the point of interest to the home location.

mric3412, 2018-09-25:
I reopen this feature request with these precisions :
- functionality achievable with simple light and inexpensive servos (no need for gimbal controller and cam gimbal / only 2x5g servos for ex. could be used for small and light models)
- no need of telemetry link to apply the Do-Set-ROI method (http://ardupilot.org/copter/docs/common-simplebgc-gimbal.html)
- HomeWP as target will be the default direction of aircraft's antenna(s) (users don't have to specify it). Managing a mobile HomeWP will be transparent for users (but suppose having a telemetry link of course)

More generally, the aim of this request is to provide the simplest, light and economic solution of airborne directional antennas easily applicable to any platform (for example, a wing <550g with battery)

Very cheap and light servos will make the job for little antennas (no need for mechanical precision here) The simpler it is, the more it will be used and the reduction of the RF impact will be good for all.

The excellent Peter's idea of using camera gimbal remains a good alternative (for powefull and heavy platform)

RickReeser, 2018-10-25:
Servo gimbals are also capable of tracking a region of interest. I think the problem might be the 360 servo: either they have a position encoder (very rare) or are rate-controlled (ie, are just motors), which won't work. I don't think there is support for encoded servos, so I guess that's what developing this feature would entail.