ArduPilot/ardupilot #9475: Set compass orientation parameters individually.

edwinhayes, 2018-09-25:
### Feature request

**Is your feature request related to a problem? Please describe.**
Configuration of compasses via a single COMPASS_ORIENTx parameter means that source code changes are required to support installation of compasses in an orientations that aren't already in the enumerated list.  This means that the list of enumerations becomes progressively longer over time as more aircraft are supported.

**Describe the solution you'd like**
Replace the single enumeration COMPASS_ORIENTx with a triplet of numerical parameters to allow the P/R/Y of each compass to be specified in degrees/radians.  Support for the old enumerated types could be effected by way of a table in the documentation side mapping enumerated types to new values, and ground station software could implement this mapping automatically in setup wizards to hide this behaviour from users if desired.

**Describe alternatives you've considered**
The alternative is pretty clear; new enumerated types for COMPASS_ORIENT are added every time a new aircraft design requires mounting of the compass in a different orientation.

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
There are currently 40 different enumerations according to the documentation.  One of which is "Yaw293Pitch68Roll180".  This doesn't seem very sane.

Pedals2Paddles, 2018-09-26:
> One of which is "Yaw293Pitch68Roll180". This doesn't seem very sane.
Tens of thousands of vehicles use that compass orientation.  It's the compass mounted in the leg of the 3DR Solo.

edwinhayes, 2018-09-26:
I didn't mean those particular values aren't sane, I know why they are there.  My question is why is this being captured in a enumeration (specifically, `enum Rotation`)?

It means that no other rotation angles can be supported without source code changes (most obviously the giant switch statement behind the implementation of `rotate()`).  Clearly `rotate()` was implemented so as to minimise the number of floating point operations used, but is this really a concern today?  As you say, there are plenty of Solo's out there, and `ROTATION_ROLL_90_PITCH_68_YAW_293 ` take 3x3 floating point multiplies anyway.

Pedals2Paddles, 2018-09-27:
Is there an actual common use case for totally custom compass orientation values?  Are we adding additional orientations to enumeration often? Or out another way...  What is worse, updating an enumeration once in a while, or turning 3 simple parameters into 9 complicated parameters?

edwinhayes, 2018-09-27:
Like I said, the enumeration is up to 40 now, so there has been at least that much demand.  But beyond that it's difficult to know, since demand may be suppressed: 

If we're designing a new aircraft, we deliberately restrict the compass placement, which may result in a suboptimal design in some other respect, because otherwise we are forced to either run a custom build solely for the purpose of supporting the compass orientation, or wait for a new stable build including the new orientation.

It's not only updating an enumeration; adding a new orientation actually requires modification to the underlying vector maths library, because of the way the `rotate()` method is implemented.  I understand the historical reason for this, but today it's horrible practice and should be fixed.

Finally, saying that it's turning 3 simple parameters into 9 complicated ones seems subjective:  I would say that a set of parameters representing Euler angles is far simpler than a mysterious enumeration, as well as being easier to document, easier to maintain, and handling every possible future configuration.

andyp1per, 2020-03-08:
Fixed by #13586

closed 2020-05-04 by tridge