ArduPilot/ardupilot #9478: Rover: servo output range not set after SERVOx_FUNCTION changed

rmackay9, 2018-09-26:
Servo output ranges are set as part of the AP_MotorsUGV::init function.  This is only called once soon after startup meaning that if a user changes a SERVOx_FUNCTION value the servo's output range will not be correct.  This could lead to odd values being sent out to the servo until the board is rebooted.

We should first verify that the issue is real by setting a second SERVOx_FUNCTION for throttle and then compare it's output to the the servo3 outputs and see if they're different.

The solution might to one of these:

- call the motors init regularly (1hz?) while the vehicle is disarmed
- add boolean values of whether various frame types have been enabled (i..e the return value of have_skid_steering()) and then regularly check if the true/false value has changed and if it has call the init() function again.  Note that this might not work in the edge case where the user is setting up a second channel for throttleLeft or throttleRight because setting up this second channel won't affect the return value of the has_skid_steering() function.

IamPete1, 2021-09-07:
Seems like the fix is just to remove this check for armed. 

https://github.com/ArduPilot/ardupilot/blob/35a93c598811fb153e0413dbcdb06b7aba6d1d47/Rover/radio.cpp#L41

bedLad, 2023-02-08:
hello is anyone working on this issue ?