ArduPilot/ardupilot #9479: Serial Port 5 no output on firmware of ChibiOS version

Gangti, 2018-09-26:
### Feature request

ArduPilot firmware on Nuttx provide Serial Port 5 for Nuttx Shell, and NSH over Mavlink in QGroundControl. But recently, I tried firmware on ChibiOS, Serial Port 5 don't have any command shell any more, and nsh over mavlink in QGroundControl have no response.

When these features come back for the firmware on ChibiOS?

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

OXINARF, 2018-09-26:
NSH won't certainly come back as that is a NuttX feature. I haven't heard of any plans to add any command shell to ChibiOS builds and honestly don't see the need for it. In any case, you can add that idea to #8109.

closed 2018-09-26 by OXINARF

OXINARF, 2018-09-26:
You can use GDB (http://ardupilot.org/dev/docs/debugging-with-gdb-on-stm32.html) or you can use printf statements that by default are sent through the USB port.