ArduPilot/ardupilot #9482: Missing dedicated AP_Motors file for Lynxmotion Hunter Vtail 400?

KisorAlaguvel, 2018-09-27:
In AP_Motors library, like it is for Tricopter (AP_MotorsTri.cpp and .h files) or many types of supported frame types, there is no .cpp and .h dedicated for Lynxmotion Hunter Vtail 400 but it finds a place in setup_motors function (in switch-case statements) in AP_MotorsMatrix.cpp.
How does it work? Where does it take frame-specific implementations when this is directly connected to hardware?

KisorAlaguvel, 2018-09-27:
I am not sure if it is a bug report or a feature request.

rmackay9, 2018-11-15:
I think the V-tail is handle by the AP_MotorsMatrix and it can be used by setting the FRAME_TYPE = 4 ("V-Tail").  I hope to update the [motor order diagrams on the wiki](http://ardupilot.org/copter/docs/connect-escs-and-motors.html#motor-order-diagrams) in the near-ish future.

rmackay9, 2018-11-15:
I'm going to close this issue because I think it's a support item and should ideally be raised [here in the Copter support forums](https://discuss.ardupilot.org/c/arducopter/copter-36).

closed 2018-11-15 by rmackay9