ArduPilot/ardupilot #9498: Copter: 3.6.0 issues list

rmackay9, 2018-10-02:
This is a list of issues to investigate for the Copter-3.6.0 release:

- Pixhawk4 can't save parameters: https://discuss.ardupilot.org/t/parameter-writing-issue-arducopter-v3-6-0-rc10-on-a-pixhawk-4/33221
- DShot causes Cube to freeze up after 10sec. Daniel_Met: https://discuss.ardupilot.org/t/copter-3-6-0-rc10-available-for-beta-testing/33057/10
- PixhackV3 - 3rd compass not found. Varun Nelakanti: https://discuss.ardupilot.org/t/copter-3-6-0-rc8-available-for-beta-testing/32374/37
- PixhackV3 - can't do fast sampling.  Roland Mayer: https://discuss.ardupilot.org/t/copter-3-6-0-rc8-available-for-beta-testing/32374/37

ChibiOS specific issues:

- Sergey: safety button not working on Pixracer: https://discuss.ardupilot.org/t/copter-3-6-0-rc11-available-for-beta-testing/33558/3
- LidarLight3 not working.  See FRED_GOEDDERT: https://discuss.ardupilot.org/t/copter-3-6-0-rc10-available-for-beta-testing/33057/12

eovnu87435ds, 2019-10-23:
Is there a discussion for the "One user's board locks up on boot-up" issue?

rmackay9, 2019-10-24:
@eovnu87435ds, I had a look around a bit but I couldn't find the discussion... i'm pretty sure it was a one-off but feel free to raise a new discussion if you're seeing something similar.

okntrkr, 2020-04-13:
https://hizliresim.com/Nb91s2
2 elements marked in the picture in the link are faulty. what can I do ?

peterbarker, 2020-11-25:
@rmackay9 are we OK to close this?

WickedShell, 2021-04-13:
I'm going to just go ahead and close this now, as there hasn't been any response from @rmackay9 about closing this. If this is wrong and we want to keep it open, but we are well and truly past 3.6 now, so I think this is stale. If you disagree @rmackay9 please reopen it :)

closed 2021-04-13 by WickedShell